semuconsulting/pyubx2 #4: Parsing bitfields, UBX-ESF-MEAS

tgalecki, 2020-12-04:
I am getting errors parsing the UBX stream. The UBX-ESF-MEAS (0x10 0x02) is not supported on the NEO-M8U-0-10
The message is not supported and the field requires getting the length from a bit field.

I'm not sure how to extract repeated fields from a bit field.

It's easy to add new messages where the length is full sized,

>         "numOsc": U1,
>         "reserved1": U2,
>         "group": (
>             "numOsc",
>             {  # repeating group * numOsc
>                 "oscId": U1,

but I' not sure valid syntax for bit slicing the fields. What's the best way to modify ubxtypes_get.py?

It does change over time so using a fixed value isn't working.

numMea is the 5MSBs at offset 6.

19:13:02  0000  B5 62 10 02 18 00 72 D8 07 00 18 18 00 00 4B FD  µb....rØ......Ký
          0010  FF 10 40 02 00 11 23 28 00 12 72 D8 07 00 03 9C  ÿ.@...#(..rØ.....
          
19:13:02  0000  B5 62 10 02 1C 00 6D D8 07 00 18 20 00 00 CD 06  µb....mØ... ..Í.
          0010  00 0E E4 FE FF 0D 03 FA FF 05 09 0B 00 0C 6D D8  ..äþÿ..úÿ.....mØ
          0020  07 00 EE 51                                      ..îQ.
          
19:13:02  0000  B5 62 10 02 18 00 D5 D8 07 00 18 18 00 00 4D FD  µb....ÕØ......Mý
          0010  FF 10 45 02 00 11 1F 28 00 12 D5 D8 07 00 CC AC  ÿ.E....(..ÕØ..Ì¬.
          
19:13:02  0000  B5 62 10 02 1C 00 D0 D8 07 00 18 20 00 00 7C 06  µb....ÐØ... ..|.
          0010  00 0E CB FE FF 0D AC F9 FF 05 09 0B 00 0C D0 D8  ..Ëþÿ.¬ùÿ.....ÐØ
          0020  07 00 F2 AE                                      ..ò®.
          
19:13:02  0000  B5 62 10 02 18 00 38 D9 07 00 18 18 00 00 4A FD  µb....8Ù......Jý
          0010  FF 10 41 02 00 11 27 28 00 12 38 D9 07 00 95 7A  ÿ.A...'(..8Ù...z.
          
19:13:02  0000  B5 62 10 02 1C 00 33 D9 07 00 18 20 00 00 0F 06  µb....3Ù... ....
          0010  00 0E 16 FE FF 0D 5B FA FF 05 0A 0B 00 0C 33 D9  ...þÿ.[úÿ.....3Ù
          0020  07 00 49 9F                                      ..I..
          
19:13:02  0000  B5 62 10 02 18 00 9C D9 07 00 18 18 00 00 4E FD  µb.....Ù......Ný
          0010  FF 10 4E 02 00 11 20 28 00 12 9C D9 07 00 67 0E  ÿ.N... (...Ù..g..
          
19:13:02  0000  B5 62 10 02 1C 00 97 D9 07 00 18 20 00 00 85 06  µb.....Ù... ....
          0010  00 0E 77 FE FF 0D E1 F9 FF 05 0A 0B 00 0C 97 D9  ..wþÿ.áùÿ......Ù
          0020  07 00 6D A4                                      ..m¤.
          
19:13:02  0000  B5 62 10 02 18 00 FF D9 07 00 18 18 00 00 4C FD  µb....ÿÙ......Lý
          0010  FF 10 3E 02 00 11 24 28 00 12 FF D9 07 00 1F 22  ÿ.>...$(..ÿÙ...".
          
19:13:02  0000  B5 62 10 02 1C 00 FA D9 07 00 18 20 00 00 92 06  µb....úÙ... ....
          0010  00 0E 61 FE FF 0D 9F F9 FF 05 0A 0B 00 0C FA D9  ..aþÿ..ùÿ.....úÙ
          0020  07 00 E8 90                                      ..è..
          
19:13:02  0000  B5 62 10 02 18 00 63 DA 07 00 18 18 00 00 47 FD  µb....cÚ......Gý
          0010  FF 10 44 02 00 11 1C 28 00 12 63 DA 07 00 E2 E4  ÿ.D....(..cÚ..âä.
          
19:13:02  0000  B5 62 10 02 1C 00 5E DA 07 00 18 20 00 00 EF 06  µb....^Ú... ..ï.
          0010  00 0E B8 FE FF 0D C8 F9 FF 05 0A 0B 00 0C 5E DA  ..¸þÿ.Èùÿ.....^Ú
          0020  07 00 8F CE                                      ...Î.
          
19:13:02  0000  B5 62 10 02 18 00 C6 DA 07 00 18 18 00 00 4A FD  µb....ÆÚ......Jý
          0010  FF 10 4E 02 00 11 21 28 00 12 C6 DA 07 00 BA 88  ÿ.N...!(..ÆÚ..º..
          
19:13:02  0000  B5 62 10 02 1C 00 C1 DA 07 00 18 20 00 00 82 06  µb....ÁÚ... ....
          0010  00 0E 5B FE FF 0D C8 F9 FF 05 09 0B 00 0C C1 DA  ..[þÿ.Èùÿ.....ÁÚ
          0020  07 00 8A D2                                      ...Ò.
          
19:13:02  0000  B5 62 10 02 18 00 2A DB 07 00 18 18 00 00 48 FD  µb....*Û......Hý
          0010  FF 10 47 02 00 11 27 28 00 12 2A DB 07 00 81 4E  ÿ.G...'(..*Û...N.
          
19:13:02  0000  B5 62 10 02 1C 00 25 DB 07 00 18 20 00 00 1B 07  µb....%Û... ....
          0010  00 0E ED FE FF 0D FA F9 FF 05 09 0B 00 0C 25 DB  ..íþÿ.úùÿ.....%Û
          0020  07 00 B2 EF 

19:05:43  0000  B5 62 10 02 1C 00 DB 25 01 00 18 20 00 00 76 02  µb....Û%... ..v.
          0010  00 0E 06 F8 FF 0D DE F7 FF 05 54 0A 00 0C DB 25  ...øÿ.Þ÷ÿ.T...Û%
          0020  01 00 3B 91

19:05:42  0000  B5 62 10 02 18 00 EE 23 01 00 18 18 00 00 E8 11  µb....î#......è.
          0010  00 10 FA 07 00 11 A1 22 00 12 EE 23 01 00 6E F9  ..ú...¡"..î#..nù.

19:05:42  0000  B5 62 10 02 1C 00 94 21 01 00 18 20 00 00 FF 05  µb.....!... ..ÿ.
          0010  00 0E F3 FE FF 0D 4D 0B 00 05 51 0A 00 0C 94 21  ..óþÿ.M...Q....!
          0020  01 00 A5 52

tgalecki, 2020-12-04:
[u-blox 8 / u-blox M8 Receiver description - Manual](https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_%28UBX-13003221%29.pdf)

semuadmin, 2020-12-04:
Hi tgalecki,

Sorry, the ESF-MEAS message was omitted from the Gen 8 payload definitions, along with a few others I'm also now aware of. I'm currently working on a significant release that will become v0.3.0-beta and this will hopefully be available on PyPi in the next week or so, once I get round to testing it on enough devices. This release will implement the 'mandatory' components of the ESF-MEAS message type but it may take me a little longer to implement the 'optional' components (the optional calibTtag element) because - as you've identified - this requires a entirely new treatment based on bitfields. Please bear with us for now - we're on it.

semuadmin, 2020-12-04:
Hi tgalecki,

I've implemented a provisional fix in the Config_Database branch via a new parse sub-function _get_esgmeas_version(), based on my interpretation of the calibTtagValid flag in the flags field, and added a couple of test cases. This interpretation may or may not be correct - are you able to test this using your device setup and confirm?

tgalecki, 2020-12-04:
Yea, I can do that in a few hours.

Is there a specific test case/setup you would want me to use?

semuadmin, 2020-12-04:
Well, my understanding is that there in effect are two flavours of ESF-MEAS payload - one where the local receiver time has been calibrated (calibTtagValid = True), and one where it hasn't ((calibTtagValid = False). I've tested both permutations using synthetic payloads (see test_parse.py), but I'll be honest and say I'm not clear how to actually apply these calibrations to a real device with an external sensor fitted, so I'm hoping you might be able to? If not, no worries.

semuadmin, 2020-12-06:
Issue should be now resolved for all 'flavours' of ESF-MEAS in v0.3.0-beta, available on PyPi.

closed 2020-12-06 by semuadmin