semuconsulting/pyubx2 #5: RXM-PMREQ

lyngklip, 2020-12-15:
Looking at the interface description for ZED-F9B I see that the message structure for RXM-PMREQ differs from pyubx2 - in the file ubxtypes_set.py, instead of 
```python
 "RXM-PMREQ": {"duration": U4, "flags": X4},
```
I would need
```python
"RXM-PMREQ": {"version": U1, "reserved0": U3, "duration": U4, "flags": X4, "wakeupSources": X4},
```
This is probably a version issue. Do you have a policy regarding different versions of interface descriptions?

lyngklip, 2020-12-15:
An additional issue (with my modification): In the UBXMessage constructor if I put wakeupSources = b'\x06' instead of b'\x06\x00\x00\x00' then the generated message is too short. I'm not pleased that an X type needs to be input as bytes - I'd prefer an integer, but now I'm just nagging you... ;-)

semuadmin, 2020-12-15:
Hi. Ironically I've been referring to the specs for ZED-F9H, ZED-F9K, ZED-F9P and ZED-F9R and they all have RXM-PMREQ defined as per my current definition, so I'd naively assumed all the other ZED devices in the same generation would follow suite. I'll have a look at the spec for ZED-F9B and make any necessary amendments. It's normally a relatively simple matter to accommodate different versions of the same message, though I do wish u-blox were a little more consistent. Bear with me for now, I'll get back to you with an TTF asap - all being well it should be in v0.3.1 before the end of the year.

In relation to your second query, I'm afraid them's the rules; if the interface specification says the type is X4 then it expects 4 bytes, if it's U4 it expects an unsigned integer, etc. I'm afraid you'll have to take that up with u-blox ;-)

semuadmin, 2020-12-15:
Just had another look at my ZED-F9* specs and I realise now there are - as you say, two different versions of the RMX-PMREQ message without any clear indication as to which applies when. Given they're both fixed length messages, I'll probably look to distinguish by payload length.

lyngklip, 2020-12-15:
I'm sorry,  I mistyped the relevant interface spec, it was ZED-F9P, I edited accordingly. Yes, the interface spec actually defines Power management request in both ways (sections 3.16.2.1 and 3.16.2.2), I don't know what the deal is with that. I needed the longer version with the wakeupSources field.

semuadmin, 2020-12-15:
Sorted, it'll be in v0.3.1-beta along with a bunch of other new ZED-F9* message types.

lyngklip, 2020-12-16:
That's beautiful. It's annoying to have to implement though? Alternatively, have two different message definitions; "PMREQ1" and "PMREQ2". Again, annoying.

semuadmin, 2020-12-16:
Well, I'm going to have to have multiple definitions in ubxtypes_set.py anyway, but I've done that elsewhere and it's no great hardship. What I try and avoid doing is having dedicated code in ubxmessage.py for individual message types, but I can see no way round that in this case. 

I gather u-blox are moving in the direction of having a 'version' attribute in all UBX message types - at which point I can introduce suitable generic processing - but as things stand there's no consistency.

closed 2020-12-16 by semuadmin

lyngklip, 2020-12-20:
You're making it easy for me. I agree with your observations on version and consistency. I have a new one for you. Coming up.