semuconsulting/pyubx2 #8: Program stuck on ubr.read()

vikram223, 2021-01-15:
Environment MACOS:
Python version 3.8.1
Code:
```
import serial
from pyubx2 import UBXReader

stream = serial.Serial('/dev/tty.usbmodem142301',38400)  # open serial port
ubr = UBXReader(stream)
print(ubr.read())
stream.close()
```
Program hangs on ubr.read(). 
Attempted to change baudrate etc.

semuadmin, 2021-01-16:
Hi - could I ask you to take a look at the bug report template and fill in as much detail as you can on the problem you're having, including the UBX device type, version of pyubx2 you're using and if possible a binary dump of the input stream. Many thanks.

https://github.com/semuconsulting/pyubx2/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

semuadmin, 2021-01-16:
Also, have you tried running the ubxstreamer.py example in the /examples folder? This should poll the device's UBX message rates and respond with a series of CFG-MSG responses.

vikram223, 2021-01-16:
Apologies, I had the validate flag set to false. I thought I tested it with both values for the validate flag, but I must've made a mistake. You're right, my data output is only nmea stream. I am getting UBX messages when I save the ubx file using u-center but I'm not seeing it directly from the serial port I guess.

I will try the ubxsteamer.py today. Thanks a lot fr the support. Please go ahead  and close the issue.

semuadmin, 2021-01-16:
No worries.

closed 2021-01-16 by semuadmin