semuconsulting/pyubx2 #10: Parse UBX-MGA-INI as SET

lyngklip, 2021-01-31:
Hello, it's me again, your nuisance.

I'd like to parse not just UBX-MGA-INI but any UBX-MGA message as mode SET.

I'd like the parser to figure out that UBX-MGA messages are always input types (as specified in the interface description) - apart from UBX-MGA-ACK of course.

When I parse a UBX-MGA-INI file I'd like the parser to use the UBX_PAYLOADS_SET dictionary rather than UBX_PAYLOADS_GET (which doesn't contain the definition for UBX-MGA-INI).

The parser decides to use mode GET in ubxmessage.py line 705 (calling UBXMessage constructor with payload).

IMO the parser should determine from the message what type of message it is - if at all possible.

I would absolutely be willing to assist in testing...

I can provide a dump of UBX-MGA messages. I obtained said dump from a ublox AssistNow online server. My intention was to analyze the dump with the pyubx2 library of which I have become dependent ;-)

semuadmin, 2021-02-01:
Hello again. Ok so not originally an intended use case - the stream and parse methods were aimed at output (i.e. GET) messages from a GNSS device - but it's a trivial change so happy to accommodate. 

The provisional changes are in the parse_inputs branch as v0.3.17.

I don't really want to get into 'auto-detecting' message mode as many input/output payload definitions are identical, but what I have done is add an optional 'mode' parameter to the UBXMessage.parse() method and the UBXReader() constructor. This will default to 0 (GET) but can be set to 1/2 if you want to stream or parse input message types (SET, POLL).

I've updated the \examples\ubxfile.py example to accommodate this new parameter and I've run your assist.ubx dump through it using mode = 1 (SET). Results copied below. Have a go yourself and let me know if this adequately addresses your issue.

**Console log from ubxfile.py:**

Enter fully qualified name of file containing binary UBX data: D:\assist.ubx
Do you want to validate the data stream (y/n)? (n) y
Message mode (0=GET (output), 1=SET (input), 2=POLL (poll)? (0) **1**
Instantiating UBXStreamer class...
Opening file D:\assist.ubx...
Starting file reader
<UBX(MGA-INI-TIME_UTC, type=16, version=0, ref=b'\x00', leapSecs=18, year=2021, month=1, day=31, hour=18, minute=34, second=26, reserved1=0, ns=551000000, tAccS=0, reserved2=0, tAccNs=500000000)>
<UBX(MGA-INI-POS_LLH, type=1, version=0, reserved1=0, lat=558392940, lon=122707060, alt=3970, posAcc=50000)>
<UBX(MGA-GPS-EPH, type=1, version=0, svId=2, reserved1=0, fitInterval=0, uraIndex=0, svHealth=0, tgd=-38, iodc=39, toc=4500, reserved2=0, af2=0, af1=-36, af0=-1228433, crs=3028, deltaN=12959, m0=-1210412793, cuc=2829, cus=1939, e=175175447, sqrtA=2701973881, toe=4500, cic=156, omega0=-1034707056, cis=227, crc=10013, i0=657312294, omega=-1061190575, omegaDot=-23335, idot=830, reserved3=0)>
...
<UBX(MGA-GAL-TIMEOFFSET, type=3, version=0, reserved1=0, a0G=-119, a1G=-39, t0G=24, wn0G=31, reserved2=0)>
<UBX(MGA-GAL-UTC, type=5, version=0, reserved1=0, a0=0, a1=0, dtLS=18, tot=0, wnt=95, wnLSF=137, dN=7, dTLSF=18, reserved2=0)>
<UBX(MGA-INI-EOP, type=48, version=0, reserved1=0, d2kRef=0, d2kMax=9496, xpP0=33586342, xpP1=14344, ypP0=385745984, ypP1=-3287, dUT1=0, ddUT1=0, reserved2=439254758650600145052798411358100931983675757265114634832462684156974167369066390015615966154413)>

**153 messages read from D:\assist.ubx.**

Closing file...
Test Complete

lyngklip, 2021-02-01:
That's beautiful. It absolutely addresses my issue. Case closed. :-)

closed 2021-02-01 by lyngklip