semuconsulting/pyubx2 #12: How to read ubx flags

djdan7, 2021-02-25:
I have a problem how to understand and read ubx flags. 
I.e. I want to read NAV-STATUS message:

raw data:
`b'\xb5b\x01\x03\x10\x00\xe0\xc8a\x18\x03\xdf\x02\x08D\x02\x00\x004S\x1e\x00\x0c\xf3'`

parsed data:
`<UBX(NAV-STATUS, iTOW=17:37:25.600000, gpsFix=3, flags=b'\xdf', fixStat=b'\x02', flags2=b'\x08', ttff=580, msss=1987380)>`

Precisely - I want to get **carrnSoln** value, which is under **flags2** (page 141 in ZEDF9P Interface Description: [https://www.u-blox.com/en/docs/UBX-18010854#%5B%7B%22num%22%3A1369%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C59.527%2C699.177%2Cnull%5D](url)

How to read `b'\x08'` and extract from it the **carrSoln**?

Thank you so much for help! :D

semuadmin, 2021-02-25:
Hi. OK so this is more a Python general knowledge question, but as you've helped me improve the product with an earlier bug report, I'll offer some pointers :-)

Two concepts you need to get to grips with here - one is **Python bitwise operations**, the other is the concept of '**endianness**' (I'm assuming you're familiar with the binary numbering system).
For the first, try Googling 'Python bitwise operations', in particular the `&` (AND) and `>>` (SHIFT) operators.
For the second, check out https://en.wikipedia.org/wiki/Endianness. UBX GNSS devices are 'little endian'.

Firstly, convert the NAV-STATUS flags2 byte to an integer using the `int.from_bytes()` function:
```
>>> int.from_bytes(b'\x08', byteorder='little')
8
```
It may be helpful to visualise this as a binary number with leading zeros - you can use the `format()` function for this (bits are numbered from right to left; bit 0 represents the '1s' and bit 7 represents the '128s')
```
>>> format(int.from_bytes(b'\x08', byteorder='little'), '#010b')
'0b00001000'
```

If you want to check whether a specific bit is set, you can perform a logical AND '&' with a 'bitmask' representing the bit(s) you're interested in - so, from the right, bit 7 is 128, bit 6 is 64, bit 5 is 32, bit 4 is 16 and so on. To get carrSoln, which is bits 6 and 7, you can AND with the bitmask 192 (64 + 128) e.g. taking various possible values of the flags2 byte:
```
>>> 0b00001000 & 192
0 (meaning neither bit 6 nor 7 is set)
>>> 0b01001000 & 192
64 (meaning bit 6 is set but not bit 7)
>>> 0b10001000 & 192
128 (meaning bit 7 is set but not bit 6)
>>> 0b11001000 & 192
192 (meaning both bits 6 and 7 are set)
```

A more common way to do this is to 'shift' the integer first to remove any bits you're not interested. So in this case, we're only interested in bits 6 and 7 so we shift the integer 6 places to the right to remove bits 0-5 using the `>>` operator, and then perform an `&` operation on the remaining bits (only this time we use 3 as the bitmask):
```
>>> 0b11001000 >> 6 & 3
3
>>> 0b01001000 >> 6 & 3
1
```
etc.

Get the drift?

FYI if you want to see this kind of thing in action in pyubx2, have a look at the `UBXMessage._get_esfmeas_version()` function.

Hope this helps.

closed 2021-02-25 by semuadmin

djdan7, 2021-02-25:
Thank you so much! Low level operations are new to me, but I will try to dig into it. Thank you! :D

semuadmin, 2021-02-25:
One of the great things about Python is that it has an interactive (or 'interpreted') console, so you can try these various operations out on the command line and get instant results - it's a very good way to get to grips with bitwise operations.

djdan7, 2021-02-26:
Thank you my Master!!! :D

I think that I got what's going on here :D I've written several functions for different bitfields and they look something like this:

```
def nav_status_flags2(self, bstr):
        bstr = int.from_bytes(bstr, byteorder='little')
        bstr = format(bstr, '#010b')
        carrSoln = int(bstr[-8:-6], 2)
        if carrSoln == 2:
            carrSoln = "FIXED"
        elif carrSoln == 1:
            carrSoln = "FLOAT"
        else:
            carrSoln = '-:-'
        return carrSoln
```

**PS.** I would like to donate you some cup of coffee for your help! :D Do you have maybe [https://www.buymeacoffee.com/](https://www.buymeacoffee.com/) profile or something similar?

semuadmin, 2021-02-26:
Whatever works for you is fine, but that's not quite the approach I was advocating for the general extraction of individual bits from UBX bitfields ('X' type attributes), and it may not work for bitfields longer than 1 byte. There's no need to convert a bitfield to an integer and then to a binary string in every case - I only did that in my example above to illustrate the underlying concepts. If you're after a generic 'bit extraction' function, I'd advise you to get to grips with the `&`, `<<` and `>>` operators; you could try something like this:

```
def get_bits(bitfield: bytes, bitmask: int) -> int:
    """
    Get integer value of specified (masked) bit(s) in a UBX bitfield attribute (type 'X')

    :param bitfield bytes: bitfield byte(s)
    :param mask int: bitmask as integer (= Σ(2**n), where n is the number of the bit)
    :return: value of masked bit(s)
    :rtype: int
    """

    i = 0
    val = int(bitfield.hex(), 16)
    while mask & 1 == 0:
        mask = mask >> 1
        i += 1
    return val >> i & mask
```

The 'bitfield' parameter is the flag you're interested in - in your case that's the `flags2` field. 'bitmask' is an integer representing the bits your interested in - you can pass this as a binary integer if that makes it easier to visualise (_remembering that bits are numbered from right to left starting with 0_) e.g. if you want bits 6 & 7 in an 8-bit field, use the bitmask 0b11000000 (decimal 192); if you want bits 3 & 4, use a bitmask of 0b00011000 (decimal 24) - see how it works?

So assuming `msg` is a parsed NAV-STATUS message, to extract carrSoln (bits 6 & 7) from the `flags2` field, you would use:

```
carrSoln = get_bits(msg.flags2, 0b11000000)
carrSoln = ["-:-", "FLOAT", "FIXED"][carrSoln]
```

I think you'll find this is a better approach for more general use but, as I say, whatever works is fine.

No donation necessary, but I appreciate the offer :-)