semuconsulting/pyubx2 #15: HNR-PVT Values seem incorrect, in relation to NAV-PVT and HNR-PVT in UCenter

NeusAap, 2021-04-16:
**Describe the bug**
Some of the values in the HNR-PVT message have completely different values after reading from UART, in comparison to the Ucenter data. (I believe all the values with number format 'I4').

The iTOW, and date values come in correctly. But the Lon, Lat, gSpeed, height, mslHeight, heading, heading accuracy etc are all incorrect values (before and after adding the supplied scaling).

I checked the number formats in ubxtypes_core.py and ubxtypes_get,py. These seem to correspond with the types given in the UBlox receiver description 8 (https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_UBX-13003221.pdf)

pyUBX2 version: 1.0.7

Relevant files (also uploaded in this issues attachments, as ZIP File):
1. UBX binary file (sorry if there are hickups in the fix, I had really bad reception): https://drive.google.com/open?id=1dmmSrNeFPoTk5itSg5kI3zCpjPnxG8dg

2. Expected values (screenshot from messages view in Ucenter): https://drive.google.com/open?id=1P1b93eRtnUc4yQ40uS_6S3ELEN3YvUrK

3. Gotten values (log output of my Python program, no scaling applied, so units might be incorrect): https://drive.google.com/open?id=1pg5W18EeQw2wr1PnIPVsajNYzoxljmHV

4. Used script to read serial stream and print log messages: https://drive.google.com/file/d/13RdHPz3kWPeIEFOuqWazbgHFCM1JJbjY/view?usp=sharing

[HNR-PVT-Problem.zip](https://github.com/semuconsulting/pyubx2/files/6324012/HNR-PVT-Problem.zip)

**To Reproduce**

in UBX-CFG-PRT: Enable UART as target, prot in: UBX, prot out: UBX, Baudrate: 921600, databits: 8, stopbits: 1, parity: None, BitORder: LSB First

in UBX-CFG-HNR: NavRate: 20 Hz

I ran the script on a Raspberry Pi 4, 4GB with python 3.7.3 and pyUBX2 1.0.7

**Expected Behavior**

I expected the HNR-PVT values parsed with pyUBX2 parser (see appendix file 3 and 4), to be identical to the UCenter values (see appendix file 2). I expected that maybe some factor difference would be noticeable, but the whole values is completely different. So only scaling would not work.

**Desktop (please complete the following information):**

 - Raspberry Pi 4, 4GB, Raspberry Pi OS version 10 (buster)
 - Kernel: Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU
 - Python 3.7.3 and pyUBX2 1.0.7
 - Windows10 for the uCenter screenshot
- UART on the raspberry Pi to retrieve data
- USB on Windows 10 to double check UART data
- Ucenter v 20.10

**GNSS/GPS Device (please complete the following information as best you can):**
- Device: U-Blox NEO-M8U
- firmware version: UDR 1.5 (latest)
- protocol: 19.20

**Additional context**

The data received by NAV-PVT, does parse correctly and correspond to the values in U-Center. I checked the source code of PyUBX2 to see if the unpacking of bytes might have been wrong, but it seems to be identical for HNR-PVT and NAV-PVT. Although the values of HNR-PVT do not correspond to U-center.

I also wanted to add that the accelerometer and gyro are not calibrated at the moment, since I don't have access to a vehicle. Although, I would think that if U center can display sensible data, that I should be able to get the same data in Python.

semuadmin, 2021-04-16:
Sorry to hear about the problem, and thanks for the comprehensive issue report. Unfortunately I don't have an HNR device to hand at the moment but I'll look into this asap and get back to you.

Few queries off the top of my head:
- Do you get the same problem with conventional (low navigation rate) message types from the same device e.g. NAV-PVT or NAV-LLH?
- Do you get the same problem if you lower the navigation and baud rate?

semuadmin, 2021-04-16:
This should be fixed in v1.0.8, uploaded to PyPi now.

closed 2021-04-16 by semuadmin

NeusAap, 2021-04-19:
Hello Semu,

Thanks for the quick response and quick fix!
This seems to have fixed the problem, the information comming in seems to make sense now!

Thanks again for such quick response and fixes!

Regards,
Mats