semuconsulting/pyubx2 #17: Wrong cfg-tmode key ids

Zedliser, 2021-04-19:
**Describe the bug**

Some of the key IDs in `ubxtypes_configdb.py` are different from ZED-F9P InterfaceDescription (UBX-18010854). Since the keys IDs are different when I am trying to send UBX-CFG-VALSET message I receive NAK messages. The keys I have found so far (might be more affected):
- CFG-TMODE-ECEF_X
- CFG-TMODE-ECEF_Y
- CFG-TMODE-ECEF_Z
- CFG-TMODE-LAT
- CFG-TMODE-LON
- CFG-TMODE-HEIGHT
- CFG-TMODE-FIXED_POS_ACC
- CFG-TMODE-SVIN_MIN_DUR
- CFG-TMODE-SVIN_ACC_LIMIT

pyubx2 version: 1.0.6

**To Reproduce**

1. Create CFG-VALSET message with one of the keys above
2. Sent it to UBlox device

**Expected Behavior**

Receive ACK message and changed device configuration according to the keys.

**Desktop (please complete the following information):**

 - The operating system you're using: Ubuntu Focal Fossa
 - The type of serial connection: UART1

**GNSS/GPS Device (please complete the following information as best you can):**

 - Device Model/Generation: u-blox ZED-F9P
 - Firmware Version: HPG 1.12
 - Protocol: 27.11
 

**Additional context**

Same with current version (1.0.8).
Values a different by a single digit, e.g. "CFG_TMODE_ECEF_X" is 0x40030003 while "CFG_TMODE_ECEF_X_HP" is 0x20030006. Currently all of tmode keys in pyubx have key ids like 0x2*, while in ubx docs some are 0x2* while others are 0x4* for some weird reason.

semuadmin, 2021-04-19:
Sorry about this, and thanks for the issue report. I'll clearly need to double check the entire list as I can see there are at least a few others that are wrong according to the spec. Leave it with me.

semuadmin, 2021-04-19:
Should now be fixed in v1.0.9. Thanks again for the issue report.

closed 2021-04-19 by semuadmin