semuconsulting/pyubx2 #35: Navigation-message decoder

Nerolf05, 2021-10-30:
**Is your feature request related to a problem? Please describe.**
I would like to suggest/contribute a new additional feature. During my time as a working student, I developed a gnss-navigation message parser for the `ubx-rxm-sfrbx`-message, based on this amazing library.
In detail, the following messages are currently supported:

- GPS - C/NAV
- GPS - L/NAV
- GLON - NAV
- GAL - I/NAV
- BDS - D1
- BDS - D2 (not completely)

The proposed decoder/parser parses the raw binary-navigation message, which is available through the `ubx-rxm-sfrbx`-message, into a readable format. So one would be able to present/store/visualize the transmitted navigation data like:

- Ephemeris
- Almancs
- Clock-parameters
- and some more.

An additional feature might be a nav-rinex exporter

#### A clear and concise description of what you want to happen.

1. Would the proposed feature be interesting for pyubx2 (if not interesting: skip 2-4)
2. One could discuss my existing code and design, if something should be changed or restructured (Feedback and suggestions for improvement would be appreciated)
3. Develop tests
4. Add the feature

**Would you be willing to assist with testing?**

Yes, I would like to help and develop tests to secure a correct decoding of the navigation messages.
Additionally, I would suggest, that it might be useful to add some ublox-recordings in the `*.ubx`-format and navigation reference data for testing the navigation message decoder.

If the request relates to a specific u-blox device that is not currently supported, would you be 
willing to assist with testing e.g. by contributing a test device to the project, providing real data dumps, or writing unittest scripts?

Unfortunately,  I don't have access to an ublox-gnss receiver supporting the `ubx-rxm-sfrbx` message at the moment. But yes, I would be willing to assist with defining tests.

Best regards

semuadmin, 2021-10-30:
Hi Nerolf05,

I noticed you were working on this! Always happy to receive suggestions for new functionality, but can I just clarify the intention and functionality here? We're presumably talking about the Receiver Management Broadcast Navigation Data Subframe message (RXM-SFRBX), which is currently defined as follows:

```
    "RXM-SFRBX": {
        "gnssId": U1,
        "svId": U1,
        "reserved0": U1,
        "freqId": U1,
        "numWords": U1,
        "chn": U1,
        "version": U1,
        "reserved1": U1,
        "group": ("numWords", {"dwrd": U4}),  # repeating group * numWords
    },
```

Do I take it that your proposed utility takes the repeating ```dwrd``` group and parses this into readable almanac and ephemera data? If so, a few observations off the top of my head:

1) How did you envisage incorporating this into the main pyubx2 ```ubxmessage.py``` module? I'm thinking the simplest method might be to implement a discrete static method e.g. ```parse_nav_subframe(msg: UBXMessage)``` which takes the parsed RXM-SRFBX UBXMessage object as an input parameter and invokes your new routines to output the relevant navigation data in an appropriate format.
2) I note you're using the ```@dataclass``` decorator, which is fine but obviously only supported in Python 3.7+ (which would mean dropping Python 3.6 support. Probably no biggie; I don't imagine many people are still using Python 3.6).
3) Have you been able to test this against a real u-blox device (ublox 9 protocol 29 or later) at all?

Nerolf05, 2021-10-30:
Hi semuadmin,

Yes, I am using the `ubx-rxm-sfrbx` in [ubx_nav_msg_parser.preprocess_ubx()](https://github.com/Nerolf05/pyubx2/blob/49ecfe510eb6bd67e394aadeb29bfdc0e92c61c9/pyubx2/pyubx_nav_msg_decoder/ubx_nav_message_parser.py#L158), which calls the corresponding `ubx_nav_msg_parser._preprocess_ubx_gnss()`, which again uses the `dwrd`- group and lifts the actual transmitted nav data, according to the payload definitions from the integration manual.

1. The static-method approach sound promising, but hereby I would suggest an interface like `parse_nav_subframe(msg: UBXMessage, nav_msg: GnssNavMsg)` as there are some inter-subframe dependencies (for instance msb- and lsb- byte parts or even three parts for BDS-D2 nav-msg), which cannot be resolved by reading one single subframe. So maybe an existing nav-msg object is filled iteration for iteration. Hereby, I have to admit, that these inter-subframe dependencies are a little-bit tricky to consider, and my approach by marking _msb_ or _lsb_ fields is probably improvable.
2. I would leave the decision to you, although using `@dataclass` reduces lines of code.
3. Yes, I tested it with a real u-blox device (ZED-F9P), more precisely with resulting ubx-records. But in the meantime I no longer have access to it.

foxittt, 2021-10-30:
Hi.
I am interested in obtaining almanac and ephemeris data from ublox.  If necessary I can test code with real ublox f9p.

Nerolf05, 2021-10-30:
Hi foxittt, 
that would be very helpful. Maybe you can also share ubx-records (tracking a specific gnss-system for the required amount of time to record the whole nav-msg). Those files would be very helpful for developing and testing.

Are there any issues sharing those files  (be aware that you would share your position as well)

foxittt, 2021-10-30:
Nerolf05, tell me which ubx-records are required. I'll try to record.

Nerolf05, 2021-10-30:
Thanks for your efforts. The required messages would be:

- `ubx-nav-pvt` (just to verify that a valid fix was achieved)
- `ubx-rxm-sfrbx`  (containing the actual nav-data)

I would guess it might be nice to have one record for each gnss-system (GPS, GAL, GLO, BDS) - especially for testing -, but one record with all systems at once should be fine as well. Note that you have to enable the respective gnss-systems.
Depending on your location, you might be able to receive BeiDou D2 (geo-stationary sats prn: {1-5})

foxittt, 2021-10-30:
How long records must be?

semuadmin, 2021-10-30:
Guys,

Very happy to see and support this level of collaboration in relation to the pyubx2 project, but I'd be interested to understand your respective use cases as this might help me better advise where this should sit in relation to the core pyubx2 modules (or whether it's actually a separate Python application which simply uses the pyubx2 library). 

What are you actually intending to do with the parsed almanac and ephemeris data once you have it?

Depending on your intended utilisation(s) (_I'm guessing this is RTK related_), might you not also need ```ubx-rxm-raw``` and/or ```ubx-rxm-rawx``` message data?

If it helps, I can happily provide PVT, SFRBX and RAWX time series data from GPS, GLONASS, BeiDou-1, Galileo and SBAS at my location using a decent ground plane, but I don't think I can get BeiDou-2. But @Nerolf05 it's obviously going to be easier if you can invest in your own device. As I say, a relatively cheap low-end NEO-M9N devkit is capable of providing the relevant raw data - just depends on whether you need higher-end capabilities like RTK, HNR or ADR out of the box.

semuadmin, 2021-10-31:
> How long records must be?

Presumably if you're looking at GPS L1 C/A you would need at least 12.5 minutes of SFRBX data (25 subframes * 30 seconds per subframe).
For Beidou D1 you would need at least 12 minutes of data (24 subframes * 30 seconds per subframe).
For GLONASS P I think it's also 12 minutes.

semuadmin, 2021-10-31:
@Nerolf05 This is just a suggestion, but having considered this a little more I'm wondering if you might actually be better off developing your navigation message parser as a separate application - say, ```pyubxrtk```? - which uses the pyubx2 library, rather than as a fork of pyubx2 itself. That way you're free to work to your own development and deployment methodology, style and release cycle.

Again, I'm not sure what your intended use case is, but if for example you have in mind some kind of RTK application using time-series ephemeris and almanac data (e.g. a Python equivalent of the [RTKLIB C library](http://www.rtklib.com/)), this is a highly specialised area which probably goes well beyond what pyubx2 was intended for i.e. a simple reader, parser and generator of individual UBX messages. If RTK _is_ an intended use case, you might also need to consider pulling in RXM-RAWX data.

Either way, happy to help in any way I can (e.g. testing, or any simple modifications or enhancements to pyubx2 which might make your life easier), and I'd be more than happy to add a reference and link to the finished navigation message parser from the pyubx2 home page.

Nerolf05, 2021-10-31:
@semuadmin the intended use case would be only to monitor the decoded nav-data. At least, I don't plan to use the decoded data for more advanced analysis, like RTK.

@foxittt yes, as already mentioned, round about 15minutes would be required.

foxittt, 2021-10-31:
@Nerolf05 Send me your e-mail for test data.
I recorded PVT, SFRBX, MEASX and RAWX for gps
My e-mail is asnmedd@gmail.com

foxittt, 2021-11-01:
> What are you actually intending to do with the parsed almanac and ephemeris data once you have it?

I am interested in analyzing data on the location of navigation satellites from different sources: ephemeris, almanac, TLE files, SP3 files.

In geodetic-class navigation receivers, ephemeris and almanac are available decoded in the corresponding messages.
The 9 series ublox receivers are probably the most affordable dual frequency multisystem receivers available.
But the data on ephemeris and almanac are available only in raw form.

semuadmin, 2021-11-02:
@Nerolf05 For what it's worth, I've created a separate branch `decode_RXM-SFRBX` which is intended to illustrate a possible approach to decoding navigation data in RXM-SFRBX in a manner that is extensible and consistent with current `pyubx2` coding conventions. **NB:** I've only spent a few hours on this so it's very much a work in progress and only partially handles GPS data at the moment, but you'll hopefully see where I'm going with it. Have a look at RELEASE_NOTES.MD and CONTRIBUTING.MD for further information and caveats. I'd welcome your thoughts given the expertise you've already acquired in this area.

Nerolf05, 2021-11-02:
Thank you very much for the efforts it looks promising, I will try to implement my code on the weekend.

semuadmin, 2021-11-02:
Here's an example of the sort of thing my provisional code in the `decode_RXM_SFRBX` branch would output
(obviously early days - bound to be some typos and would need rigorous testing, but you get the drift):

[pygpsdata-20211102183339.log](https://github.com/semuconsulting/pyubx2/files/7463253/pygpsdata-20211102183339.log)

Nerolf05, 2021-11-06:
Some remarks/questions on `decode_RXM_SFRBX`:

- Do you intend to include the decode_nav_data functionality directly into the `UBXMessage` class, so to say on a Message-level -> This approach might lead to some difficulties for intersubframe-dependencies like BDS-D1-subframe 2/3 for `toe` parameter
[image]

- Another idea might be to implement a `ubx_lift_nav_data` - function, such that the actual nav_data_parser could be used for **every** GnssReceiver supporting raw-nav-data output, and not only ublox-devices, like 
- The idea, as presented in `ubxnavdecode.py` would be to decode fields, such that the user can struct/group them as required?

semuadmin, 2021-11-07:
> Some remarks/questions on `decode_RXM_SFRBX`:
> 
>     * Do you intend to include the decode_nav_data functionality directly into the `UBXMessage` class, so to say on a Message-level -> This approach might lead to some difficulties for intersubframe-dependencies like BDS-D1-subframe 2/3 for `toe` parameter
>
>     * The idea, as presented in ubxnavdecode.py would be to decode fields, such that the user can struct/group them as required?

Hi @Nerolf05,

So my position is that I want `pyubx2` to remain a single-pass UBX parser - that is, parse each incoming UBX message into a single `UBXMessage` object. The `decode_RXM_SFRBX` fork was intended as an illustration of how you might decode raw nav data on this basis. This approach works OK for GPS L1 subframes but - as you say - it wouldn't work if individual attributes are spread across multiple subframes arriving as separate UBX messages, as appears to be the case for the BDS-D1 `toe` msb/lsb parameters.

I must confess that I haven't studied the other GNSS subframe specifications in detail yet. Do you happen to know if there are similar issues with decoding Galileo, GLONASS or other augmented GNSS subframes, or is BDS-D1 `toe` the only example of data being split across subframes in this way?

If decoding nav data requires the aggregation of dwrds from multiple incoming UBX messages, then (*as I mentioned before*) I would see that as a separate application - **not** part of the core `pyubx2` library. It simply adds too much complexity and maintenance overhead to what is currently a fairly lightweight and agile module.

If people think there's value in continuing to develop my `ubxnavdecode.py` module to accommodate the other GNSS systems, and let the end user sort out some kind of 'fudge' for the BDS-1 `toe` msb/lsb, then I'm happy to receive contributions in this space.

But I'm inclined to think that the best and most flexible approach is to develop this as a separate application (e.g. `pygnssdecode` or similar) using the `pyubx2` library for basic UBX message parsing. This gives you the flexibility to adopt your own programming style and accommodate some of the additional functionality you've alluded to above, like RINEX conversion or decoding nav data from non-UBX receivers. I'd strongly encourage you to continue the work you've done on this basis, rather than as a fork of  `pyubx2`. I will more than happily add a link and reference to it from the `pyubx2` home page once it's ready.

Hope this helps. Let me know what you think.

semuadmin, 2021-11-07:
@Nerolf05 As an afterthought, you might want to take a look at https://github.com/semuconsulting/PyGPSClient and see if you think there's any mileage in developing your nav data decode functionality as an extension of that instead? It's intended as a more general purpose GNSS utility application.

PyGPSClient already does some fairly simple processing on a 'time-series' basis. For example, it aggregates data from multiple NMEA GSV messages to produce the satellite graph and position charts.

Just a suggestion???

Nerolf05, 2021-11-07:
@semuadmin

> I must confess that I haven't studied the other GNSS subframe specifications in detail yet. Do you happen to know if there are similar issues with decoding Galileo, GLONASS or other augmented GNSS subframes, or is BDS-D1 toe the only example of data being split across subframes in this way?

If you have a look at the [actual payload-definitions](https://github.com/Nerolf05/pyubx2/blob/master/pyubx2/pyubx_nav_msg_decoder/navmsg_payload.py), as far as I implemented them, there are several dependencies for the GAL-Almanacs and BeiDou in general, but in especially D2-Nav of geo-sats, which would lead to quite a number of special-cases.

Considering the point, that `pyubx2` remains a UBX-parser for single messages, I would  agree that a separate module makes more sense, since decoding aggregated messages is at another processing-level. Moreover, it would make more sense for the intended modularity, as individual receivers could be added quite easy.

I would be very happy if it ended up being used by `PyGPSClient`. 
So I could imagine that, for example, one specific (tracked) satellite can be selected and all transmitted nav-dat {ephemeris, clock-data, health, complete Almanac} is presented. Or a Rinex-Navigation file might be generated,

semuadmin, 2021-11-07:
Ok I think we're in agreement that it probably sits outside `pyubx2`, in which case I'll close this issue for now (and drop my `decode_RXM_SFRBX` branch), but I'll flag it up as a possible future enhancement to `PyGPSClient`.

In the meantime, I recommend you continue to develop as a stand-alone application - say, `pygnssdecode` or similar - and drop me a line if I can be of any assistance with testing or whatever.

Great work so far!

closed 2021-11-07 by semuadmin