semuconsulting/pyubx2 #44: MGA messages cannot be parsed

Hadatko, 2021-11-22:
**Describe the bug**

Hello. I have issue to parse MGA messages.
In your comments is mentioned:     
# ***************************************************************
    # Multiple GNSS Assistance messages
    # These need special handling as MSGIDs alone are not unique;
    # message identity is determined by 'type' attribute in payload
    # ***************************************************************
Are you planing to support this somehow? I can help but i wan't do PR before small talk. 

My current idea: 
Change UBX_MSGIDS from:
UBX_MSGIDS={key:value} to UBX_MSGIDS={key:{key2:value}} 
for unique messages it would looks like UBX_MSGIDS={key:{"":value}}
key2 would be type in MGA messages. What would you say about that?

Please specify the pyubx2 version (`>>> pyubx2.version`) and, where possible, include:
current version
[image]

**To Reproduce**

Steps to reproduce the behaviour:
`pyubx2.UBXReader.parse(b'\xb5\x62\x13\x40\x18\x00\x10\x00\x00\x12\xe5\x07\x0b\x16\x09\x3A\x3b\x00\x80\x73\x3b\x0D\x0f\x00\x00\x00\x00\x00\x00\x00\x62\x3f')`

**Expected Behavior**

Parse message successfuly.

Hadatko, 2021-11-22:
The issue looks actually a bit different. So i will close this ticket.

closed 2021-11-22 by Hadatko

Hadatko, 2021-11-22:
I found the issue. I didn't know that i need specify msgmode, otherwise default GET is used.
`pyubx2.UBXReader.parse(b'\xb5\x62\x13\x40\x18\x00\x10\x00\x00\x12\xe5\x07\x0b\x16\x09\x3A\x3b\x00\x80\x73\x3b\x0D\x0f\x00\x00\x00\x00\x00\x00\x00\x62\x3f',msgmode=1)`

semuadmin, 2021-11-22:
Hi @Hadatko,

Yes. For various technical reasons, pyubx2 distinguishes between between Output (GET), Input (SET) and Poll (Poll) message types via the message mode parameter. MGA messages are (mostly) INPUT messages - that is, they are sent FROM an AssistNow service TO the receiver, so you require a message mode of SET (0x01) to parse using the appropriate payload definition. This is highlighted in the README with explicit examples, but it does still catch people out. Glad the problem is sorted.