semuconsulting/pyubx2 #46: UBX-ESF-STATUS messages seem incorrect

pezosanta, 2022-01-04:
I have a **NEO-M8U-0-10** module and want to monitor the status of the sensor fusion calibration process with a python script running on a Windows 10 desktop (with USB serial connection). 

However, the parsed **ESF-STATUS** messages (shown below) differ from the UBX-ESF-STATUS messages shown in the documentation.

`<UBX(ESF-STATUS, iTOW=17:20:34.350000, version=2, reserved1=36, reserved2=2, reserved3=0, status=0, reserved4=0, reserved5=0, numCh=6, type_01=5, used_01=0, ready_01=1, calibStatus_01=0, timeStatus_01=1, freq_01=0, badMeas_01=1, badTTag_01=0, missingMeas_01=1, noisyMeas_01=1, type_02=4, used_02=0, ready_02=0, calibStatus_02=2, timeStatus_02=2, freq_02=142, badMeas_02=0, badTTag_02=0, missingMeas_02=1, noisyMeas_02=0, type_03=10, used_03=0, ready_03=0, calibStatus_03=0, timeStatus_03=0, freq_03=4, badMeas_03=0, badTTag_03=1, missingMeas_03=0, noisyMeas_03=1, type_04=0, used_04=0, ready_04=0, calibStatus_04=1, timeStatus_04=0, freq_04=10, badMeas_04=0, badTTag_04=0, missingMeas_04=0, noisyMeas_04=0, type_05=18, used_05=0, ready_05=1, calibStatus_05=0, timeStatus_05=1, freq_05=0, badMeas_05=0, badTTag_05=0, missingMeas_05=0, noisyMeas_05=0, type_06=0, used_06=0, ready_06=0, calibStatus_06=0, timeStatus_06=0, freq_06=0, badMeas_06=0, badTTag_06=0, missingMeas_06=0, noisyMeas_06=0)>`

In the message payload:
- the _fusionMode_ and _numSens_ fields are missing
- there are not only _reserved1_ and _reserved2_ fields (like in the docs) but _reserved3_, _reserved4_ and _reserved5_ fields as well
- different from the doc, there is a _status_ fileld and a _numCh_ field as well
- according to the docs and the U-Center application, the _type_ (sensor data type) fields are supposed to be either 5 (z-axis gyroscope), 13 (y-axis gyroscope), 14 (x-axis gyroscope), 16 (x-axis accelerometer), 17(y-axis accelerometer), 18 (z-axis accelerometer), but in the parsed messages _type_ fields are mainly 0 (none, data field contains no data) or 4 (reserved)

To reproduce this behaviour use the following code snippet:
`
from serial import Serial
from pyubx2 import UBXReader

stream = Serial('COM5', 115200, timeout=3)
ubr = UBXReader(stream)

while True:
    (raw_data, parsed_data) = ubr.read()

    if parsed_data.identity == 'ESF-STATUS':
        print(parsed_data)
`

semuadmin, 2022-01-04:
Hi @pezosanta ,

Thanks for bringing this to my attention, and sorry for the trouble you've been having.

I've just reviewed the documentation for ESF-STATUS from various generations of u-blox device. Some of these issues appear to be cosmetic changes from older documentation e.g. the `status` field is equivalent to `fusionMode`, `numCh` is equivalent to `numSens`, and the various `reserved` fields have different names in older documentation but occupy the same payload positions. There does however appear to be a genuine error in my definition of the `sensStatus2` bitfield, which I have as `X2` but which should be `X1`. I'm guessing this accounts for the payload discrepancies you highlight above, at least for the 2nd and subsequent sensor readings.

I believe the correct definition should be as follows:

```
    "ESF-STATUS": {
        "iTOW": U4,
        "version": U1,
        "reserved0": U7,
        "fusionMode": U1,
        "reserved1": U2,
        "numSens": U1,
        "group": (
            "numSens",
            {  # repeating group * numSens
                "sensStatus1": (
                    X1,
                    {
                        "type": U6,
                        "used": U1,
                        "ready": U1,
                    },
                ),
                "sensStatus2": (
                    X1,
                    {
                        "calibStatus": U2,
                        "timeStatus": U2,
                    },
                ),
                "freq": U1,
                "faults": (
                    X1,
                    {
                        "badMeas": U1,
                        "badTTag": U1,
                        "missingMeas": U1,
                        "noisyMeas": U1,
                    },
                ),
            },
        ),
    },
```

I'll get onto this asap.

pezosanta, 2022-01-20:
Hi @semuadmin,

Sorry for the late reply, I was focusing on my other job tasks and had some das off as well.

Thank You for your amazingly rapid bug-fix ! :) Now the UBX-ESF-STATUS messages seem to be correct.

`<UBX(ESF-STATUS, iTOW=10:20:42.950000, version=2, reserved0=548, fusionMode=0, reserved1=0, numSens=6, type_01=5, used_01=0, ready_01=1, calibStatus_01=0, timeStatus_01=1, freq_01=10, badMeas_01=0, badTTag_01=0, missingMeas_01=0, noisyMeas_01=0, type_02=13, used_02=0, ready_02=1, calibStatus_02=0, timeStatus_02=1, freq_02=10, badMeas_02=0, badTTag_02=0, missingMeas_02=0, noisyMeas_02=0, type_03=14, used_03=0, ready_03=1, calibStatus_03=0, timeStatus_03=1, freq_03=10, badMeas_03=0, badTTag_03=0, missingMeas_03=0, noisyMeas_03=0, type_04=16, used_04=0, ready_04=1, calibStatus_04=0, timeStatus_04=1, freq_04=10, badMeas_04=0, badTTag_04=0, missingMeas_04=0, noisyMeas_04=0, type_05=17, used_05=0, ready_05=1, calibStatus_05=0, timeStatus_05=1, freq_05=10, badMeas_05=0, badTTag_05=0, 
missingMeas_05=0, noisyMeas_05=0, type_06=18, used_06=0, ready_06=1, calibStatus_06=0, timeStatus_06=1, freq_06=10, badMeas_06=0, badTTag_06=0, missingMeas_06=0, noisyMeas_06=0)>`

However, I have some questions (I'm new to the Ublox universe) about
1. How to check current UBX-CFG configs and how to modify them if needed
2. Whether you have any example codes of applying ["Save On Shutdown (SOS) Feature"](https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_UBX-13003221.pdf#page=26&zoom=100,0,521) (which requires a sequence of configuration actions). This would be useful to save the calibrated ESF (IMU) parameters between receiver restarts. However, the [documentation](https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_UBX-13003221.pdf#page=152) says that this can be achieved with **SOS** or if **non-volatile storage is available**. So I'm little bit confused whether using SOS is necessary in my case (**NEO-M8U-0-10 module**) or not.

If you have some spare time to help me out, should I create a new "Issue" with my questions, or reach you out on another platform (e.g. email) instead?

semuadmin, 2022-01-20:
Hi @pezosanta

Your latest questions should probably be raised on the u-blox support forum https://portal.u-blox.com/s/. Please appreciate that I don't officially represent u-blox and the work on `pyubx2` and its companion libraries `pygpsclient` and `pynmeagps` are done on an entirely unpaid voluntary basis (though you are welcome to make a [small donation](https://github.com/semuconsulting/pyubx2#author) if you find the library useful and wish to encourage its future development and support!).

Having said that, if it's a Python 3 implementation you're after, the official u-blox support forum will probably point you back in my direction :-)

`pyubx2` and the companion `PyGPSClient` GUI fully support "UBX-CFG" configuration on all u-blox GNSS devices from Generation 6 onwards and you'll find some implementation examples in the `\examples` folder. **Note that a significant change was introduced at Generation 9** (NEO-M9S, ZED-F9S, etc.) - a new Configuration Database was introduced and all configuration changes are made via the CFG-VALSET/GET/DEL messages rather than the earlier UBX-CFG messages. The UBX-CFG messages will continue to work for now but are deprecated and will be removed in future device generations. If you're using an older pre-Gen9 device (_such as your NEO-M8U_), the legacy UBX-CFG messages are still supported. See https://github.com/semuconsulting/pyubx2#configinterface for further details.

Depending on your needs, you may find the PyGPSClient GUI easier to use than the command line. It supports most of the common legacy UBX-CFG messages and a full range of the newer CFG-VALSET/GET/DEL messages. It also allows you to store user-defined command sequences that you may want to use repeatedly. See https://github.com/semuconsulting/PyGPSClient#userdefined for further details.

In terms of Save on Shutdown,

commit 60d78652 (2022-01-04): update ESF-STATUS definition Fixes #46
closed 2022-01-20 by semuadmin