semuconsulting/pyubx2 #49: pyubx2.exceptions.UBXParseError: Message checksum b'\xce\xe9' invalid - should be b'j\xf9'

GangulliGanesh, 2022-01-08:
**Describe the bug**

A clear and concise description of what the bug is.

Please specify the pyubx2 version (`>>> pyubx2.version`) and, where possible, include:
 - The error message and full traceback.
 - A binary / hexadecimal dump of the UBX data stream (e.g. from PuTTY, screen or u-center).

**To Reproduce**

Steps to reproduce the behaviour:
1. Any relevant device configuration (if other than factory defaults).
2. Any causal UBX command input(s).

**Expected Behavior**

A clear and concise description of what you expected to happen.

**Desktop (please complete the following information):**

 - The operating system you're using [e.g. Windows 10, MacOS Big Sur, Ubuntu Bionic]
 - The type of serial connection [e.g. USB, UART1, I2C]

**GNSS/GPS Device (please complete the following information as best you can):**

 - Device Model/Generation: [e.g. u-blox NEO-9M]
 - Firmware Version: [e.g. SPG 4.03]
 - Protocol: [e.g. 32.00]
 
This information is typically output by the device at startup via a series of NMEA TXT messages. 
It can also be found by polling the device with a UBX MON-VER message. If you're using the 
PyGPSClient GUI, a screenshot of the UBXConfig window should suffice.

**Additional context**

Add any other context about the problem here.

GangulliGanesh, 2022-01-08:
pyubx 1.2.3

code:
stream = serial.Serial('/dev/ttyACM0', 9600, timeout=3)
stream.flushInput()
time_now  = datetime.datetime.now().strftime('%m_%d_%Y_%H_%M_%S')
print(time_now)
file_name = time_now+'.ubx'
open(file_name,'wb')
while True:
   ubr = UBXReader(stream)
   (raw_data, parsed_data) = ubr.read()
   file1 = open(file_name, 'ab')
   file1.write(raw_data)
   file1.close()

semuadmin, 2022-01-08:
Hi @GangulliGanesh,

The idea of the issue report template is that you fill in the necessary details to assist me in diagnosing the problem, but you haven't actually filled in any of the information e.g. what device or platform you're using, or provided an extract of the binary (raw) data stream you're attempting to parse? Without such information it's going to be difficult for me to investigate the issue you're having.

I gather from your code snippet that what you're attempting to do here is read the incoming UBX data stream from (presumably) a u-blox GNSS device and write the raw (binary) data to a logfile, but if you're not parsing the UBX messages you don't actually need pyubx2 to do this - you could simply open a terminal on the prescribed serial port and pipe the output directly to a logfile e.g. using cat (or screen) on Linux/MacOS or PuTTY on WIndows:

> cat /dev/ttyACM0 > logfile.ubx

But assuming you do want to parse UBX messages at some point and you've installed pyubx2 using pip, I would recommend you try using the supplied command line tool `ubxdump` to parse your device's data stream and check that this works OK first, e.g.:

> ubxdump port=/dev/ttyACM0 timeout=3

Try this first and, if it works OK, get back to me with a screen shot of the output. Cheers.

closed 2022-01-08 by semuadmin