semuconsulting/pyubx2 #53: Load config

natuition, 2022-02-03:
Hi, 
I made this request to you in order to know if it was possible to load a ublox configuration on an f9p with your library.
For this I have a file generated by u-center.
My goal would be to be able to load a ublox configuration through a python script without a user interface.

Thank you in advance for your answers, cordially, Vincent LAMBERT.

semuadmin, 2022-02-03:
Hi @natuition,

The pyubx2 library fully supports F9P devices and is capable of sending or receiving any of the available configuration command sets. When you say you want to be able to load a configuration 'without a user interface', I presume you mean from the shell or Python command line?

However, I'm not clear what you mean by 'you have a file generated by u-center'? What kind of file? What exactly are you trying to achieve?

If you can explain exactly what you want to configure, I may be able to assist further.

natuition, 2022-02-03:
Yes, I mean from the shell or the Python command line.

My goal was to take the configuration of an F9P which is ideal in my case and to be able to replicate it on new cards. Make a backup of this functional config then be able to deploy it.

For example I set the ports, baudrate and the message on different ports.

semuadmin, 2022-02-03:
Hi.

OK so replicating the complete firmware configuration from one device to another is non-trivial. You can use the UBX UPD-SOS command to backup and restore a firmware image, but that's not something to be undertaken lightly as you risk bricking the device. If you want to replicate individual configuration parameters like port, protocol and message rates, I would be inclined to poll your 'source' device using the CFG-VALGET message, and then send the same payload to your target device(s) via the corresponding CFG-VALSET command. You can poll/set up to 64 parameters at a time. You could save this as a script and run it on demand. 

There is an illustrated example of this process in the `examples\ubxconfidb.py` module. 

Hope this helps.

semuadmin, 2022-02-03:
I'm going to close this for now, but if you need guidance on how to set up a given configuration on an F9P device using (for example) CFG-VALSET commands, you can always post a general query to the u-blox support forum at:

https://u-blox.force.com/s/

closed 2022-02-03 by semuadmin