semuconsulting/pyubx2 #55: High precision message truncation

TwardzikTomas, 2022-02-09:
**Description**
I think there is an issue with the parsing of High precision messages in UBX protocol. According to documentation, the the high precision part is supposed to introduce 2 more digits of precision (1e-9) in comparison with normal precision messages (1e-7). However when I look at the high precision part of the data obtained from puubx2 reader, all of the values are truncated to one digit in 1e-8. I suspect there might be some rounding happening in here.

I am using python 3.9 with the latest 1.2.4.

**To Reproduce**
Steps to reproduce the behavior:
Happens always with NAV-HPPOSLLH message type.

**Desktop (please complete the following information):**
 - Ubuntu 20.04 LTS
 - USB connection

**GNSS/GPS Device**
 - Device Model/Generation: u-blox ZED F9P
 - Firmware Version: HPG 1.30

semuadmin, 2022-02-09:
Hi @TwardzikTomas ,

Sorry to hear you're having problems with `pyubx2`.

UBXMessage rounds scaled attributes (like lat/lon) to 8 decimal places _after applying the scaling factor_, so there may be an issue with rounding here, but can I just clarify your expectation? Are you saying you're only getting lat/lon to a _single_ decimal place in the NAV-HPPOSLLH message, because that's certainly not consistent with our testing, though it's possible we may have missed something?

Can you provide an explicit example of raw and parsed (UBXMessage) data you're seeing from `pyubx2`, vs what you're expecting (_you can redact the 2 most significant digits if you prefer not to disclose them, or send me a log privately_)?

High Precision devices like the F9P do indeed provide an additional two significant digits of accuracy (9 rather than 7), but that's not the same as saying they are accurate to 9 decimal places. 9 decimal places implies µm accuracy. My understanding was that u-blox HP devices like the F9P only provide 'centimeter level' accuracy so 6 decimal places should in theory be ample - I round to 8 dp to provide some leeway. But, as I say, we may have overlooked something.

semuadmin, 2022-02-11:
Hi @TwardzikTomas,

Are you able to provide a sample input / output illustrating the issue you're having with NAV-HPPOSLLH messages? More than happy to investigate, but we need a little more information. Thanks.

TwardzikTomas, 2022-02-11:
Hello,
there might have been some misunderstanding. Its just the fact that High precision part of the message is always rounded to only one digit, but its probably as you have said, a leeway. Normal precision message in intect and works correctly. You have correctly mentioned that Ublox F9P is has centimeter accuracy. I probably dont understand the difference between the accuracy and significant digits, however I checked that values at 9th decimal place should refer to roughly mm displacements. Thats why you rounded at 8th?

semuadmin, 2022-02-11:
Sorry to labour this but I'm still not clear on the issue. You say the '_high precision part of the message is always rounded to only one digit_'. If you're referring to the value of, say, `latHp` or `lonHp` in the NAV-HPPOSLLH message then the value should be rounded to 8 (_not 1_) decimal places - if this is NOT what you're seeing, then there may indeed be an issue.

**This would be much easier to diagnose** if you could simply cut-and-paste a copy of the actual NAV-HPPOSLLH message you're seeing (ideally both raw and parsed). If it contains a location you'd prefer not to disclose, then by all means redact the 2 most significant digits after cutting and pasting (so **.12345678 rather than 51.12345678, for example). Thanks.

semuadmin, 2022-02-15:
@TwardzikTomas I'm going to close this issue for now, but if you continue to have problems with parsing high precision data, do please raise another issue _with specific examples of_ :

1. the data stream you're parsing
2. the outputs you're expecting
3. the outputs you're actually seeing

Thanks

commit f3edef2c (2022-03-01): Merge pull request #57 from semuconsulting/amend-scale-rounding

Amend scale rounding Fixes #55
closed 2022-02-15 by semuadmin