semuconsulting/pyubx2 #60: gpxtracker.py not working

kikislater, 2022-03-03:
**Describe the bug**

VALCKSUM import error

Please specify the pyubx2 version (`>>> pyubx2.version`) and, where possible, include:
```
>>> pyubx2.version
'1.2.6'
```

```
 $ python gpxtracker.py 
Traceback (most recent call last):
  File "/home/sylvain/SIG/RTK/gpxtracker.py", line 21, in <module>
    from pyubx2.ubxreader import UBXReader, VALCKSUM
ImportError: cannot import name 'VALCKSUM' from 'pyubx2.ubxreader' (/home/sylvain/.local/lib/python3.10/site-packages/pyubx2/ubxreader.py)
**To Reproduce**
```

Steps to reproduce the behaviour:
```
Python 3.10.2 (main, Jan 15 2022, 19:56:27) [GCC 11.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyubx2.ubxreader import UBXReader, VALCKSUM
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'VALCKSUM' from 'pyubx2.ubxreader' (/home/sylvain/.local/lib/python3.10/site-packages/pyubx2/ubxreader.py)
```

**Expected Behavior**

import correct python module

**Desktop (please complete the following information):**
```
██████████████████  ████████   sylvain@T480s 
██████████████████  ████████   ------------- 
██████████████████  ████████   OS: Manjaro Linux x86_64 
██████████████████  ████████   Host: 20L7S0XM00 ThinkPad T480s 
████████            ████████   Kernel: 5.10.102-1-MANJARO 
████████  ████████  ████████   Uptime: 3 days, 16 hours, 11 mins 
████████  ████████  ████████   Packages: 2240 (pacman), 1 (pkg) 
████████  ████████  ████████   Shell: bash 5.1.16 
████████  ████████  ████████   Resolution: 1920x1080 
████████  ████████  ████████   DE: MATE 
████████  ████████  ████████   WM: Metacity (Marco) 
████████  ████████  ████████   WM Theme: Matcha-dark-sea 
████████  ████████  ████████   Theme: Matcha-dark-azul 
████████  ████████  ████████   Icons: Papirus-Dark 
                               Terminal: mate-terminal 
                               Terminal Font: Monospace 10 
                               CPU: Intel i7-8550U (8) @ 4.000GHz 
                               GPU: Intel UHD Graphics 620 
                               Memory: 6378MiB / 15769MiB 
```

**GNSS/GPS Device (please complete the following information as best you can):**

 - Device Model/Generation: u-blox F9P
 - Firmware Version: 1.13
 - Protocol: ubx file

closed 2022-03-03 by semuadmin

semuadmin, 2022-03-03:
Hi @kikislater Sorry about that, and thanks for the report. Fixed now, but as it only affects example code I won't create another 'pyubx2' release at this point. For the fixed version, please download the modified `gpxtracker.py` from GitHub.

semuadmin, 2022-03-03:
As a suggestion, try amending `gpxtracker.py` as follows:

Change the line:

```python
for (_, msg) in self._ubxreader:  # invokes iterator method
```

to this:

```python
for (_, msg) in self._ubxreader.iterate():  # invokes iterator method with exception handling wrapper
```

... and run it again. It will still raise an invalid checksum error for any corrupt messages, but should continue with the remainder of the (valid) data stream. Let me know how you get on.

commit e3cff788 (2022-03-03): Update gpxtracker example Fixes #60