semuconsulting/pyubx2 #61: Possible to Brick GPS Module?

dHofmeister, 2022-03-18:
Dear Semu,

I was wondering if it is in some way possible to use this library in some way that could lead to bricking a drotek zed-f9p gps module? 
I currently have 2 "dead" GPS modules that do not show up as /dev/ttyACM* devices anymore and on windows they show as "USB device not recognised". 

The script that i'm running starts with running 3 threads, one thread pushes a config every 5 seconds, the second thread sends/polls the config every 2 seconds, and the third thread reads the serial buffer and confirms that the received config matches the sent config. 

Once that is confirmed, these threads exit and a new serial read thread is started as a main thread, as the only thread. 

Could I have bricked the GPS device with such a setup? Is there any other way I could have bricked the device through software/serial? Is the gps module truly bricked or could I somehow hard-reset it? 

greetings.

semuadmin, 2022-03-19:
Hi @dHofmeister 

I am not aware of any circumstance in which use of this library would 'brick' a u-blox device to the extent that it was no longer recognisable as a USB device, and we have never had any other reports to this effect.

Have you contacted the manufacturer(s) of your module(s) for their advice on how to reset them?

If you can you post your complete Python script, we can take a look at it, but I think it highly unlikely that any Python script would cause permanent damage unless you've managed to  upload corrupted firmware or there was some kind of voltage surge on your USB bus (i.e. hardware issue).

dHofmeister, 2022-03-19:
Dear Semu, 

Thank you for your time. You can find the code below. 

I am also sceptical that such a thing could be possible, but I wanted to double check it with you. I took the leap of asking you as I saw that it was mentioned in another python UBX library. https://github.com/mayeranalytics/pyUBX

> "The pyUBX software is GPL 3.0 licensed. The software is provided "as-is". Use it carefully or you might brick your device!!!"

But I also am thinking that this could be a pure hardware issue. 

Greetings and thank you very much for your time energy.

```
import time

import threading

from serial import Serial
from pyubx2 import UBXReader, UBXMessage, ERR_IGNORE, VALCKSUM, UBX_PROTOCOL, SET_LAYER_RAM, SET_LAYER_FLASH, \
    POLL_LAYER_RAM, POLL_LAYER_FLASH, TXN_START, TXN_ONGOING, TXN_COMMIT, UBX_MSGIDS

class GPS():
    def __init__(self, **kwargs):
        self.port = "/dev/ttyACM0"
        self.serial_port = None
        self.ubx_reader = None
        self.is_configured = False
        self.ram_verified = False
        self.flash_verified = False
        # MOVING BASE MODE
        # RTCM OUTPUT ON UART2
        self.baudrate = 460800
        self.configuration = ([
            ("CFG_UART1_ENABLED", 0),
            ("CFG_UART2_ENABLED", 1),
            ("CFG_USB_ENABLED", 0),
            ("CFG_SPI_ENABLED", 0),
            ("CFG_I2C_ENABLED", 0),

            # # RATE / FREQUENCY
            ("CFG_RATE_MEAS", 200),
            ("CFG_RATE_NAV", 1),

            ## UART 1
            # BAUDRATE
            ("CFG_UART1_BAUDRATE", int(self.baudrate)),

            # INPUT
            ("CFG_UART1INPROT_NMEA", 0),
            ("CFG_UART1INPROT_RTCM3X", 0),
            ("CFG_UART1INPROT_UBX", 0),

            # OUTPUT
            ("CFG_UART1OUTPROT_NMEA", 0),
            ("CFG_UART1OUTPROT_RTCM3X", 0),
            ("CFG_UART1OUTPROT_UBX", 0),

            ## UART 2
            # BAUDRATE
            ("CFG_UART2_BAUDRATE", int(self.baudrate)),

            # INPUT
            ("CFG_UART2INPROT_NMEA", 0),
            ("CFG_UART2INPROT_RTCM3X", 0),
            ("CFG_UART2INPROT_UBX", 0),

            # OUTPUT
            ("CFG_UART2OUTPROT_NMEA", 0),
            ("CFG_UART2OUTPROT_RTCM3X", 1),
            ("CFG_UART2OUTPROT_UBX", 0),
[... 99 lines cut ...]
        # Match the returned config with the desired config
        for entry in self.configuration:
            configured = configured and getattr(msg, entry[0]) == entry[1]
        return configured

    def run(self):
        while True:
            try:
                msg = self.ubx_reader.read()
            except:
                continue

    def connect(self):
        self.serial_port = Serial(self.port, self.baudrate, timeout=1)
        self.ubx_reader = UBXReader(self.serial_port, quitonerror=ERR_IGNORE, validate=VALCKSUM, protfilter=UBX_PROTOCOL)

    def get_keys(self) -> list:
        l = []
        for t in self.configuration:
            l.append(t[0])
        return l

if __name__ == '__main__':
    gps = GPS()
```

semuadmin, 2022-03-20:
I'm away from the office for a few days so this is a quick response from a mobile phone. At first glance I can see a number of potential issues with your code, though - as I say - I'd be surprised if these have resulted in permanent damage. I can't comment on other open-source libraries.

1) the CFG-VALSET command can accommodate up to 64 key/value pairs in a single message so there's absolutely no need to mess with timed loops or transactional updates here. Just send a single non-transactional CFG-VALSET message containing the required config key/val pairs and wait for an ACK-ACK response before proceeding further.
2) Same goes for the CFG-VALGET poll.
3) You can OR multiple memory layers in the CFG-VALSET command e.g. ram | bar | flash BUT...
4) ... it's prudent to apply new configurations in volatile memory first (ram) before committing them to non-volatile (flash). That way, a simple power cycle will reset the device to its previous configuration. Avoid updating flash until you're absolutely confident your updates are correct.
5) your baud rate is unnecessarily high. Although 460800 is the theoretical maximum for F9P devices, some uart peripherals may struggle at this rate and it may be safer to use the default baud rate of 38400 for simple configuration updates.

It's possible that inappropriate use of transactional updates to flash memory may have left your device in an indeterminate state, so I recommend you consult the manufacturer for advice on how to reset it to the default settings - your board may support a pin-driven hardware reset.

Hope this helps.

dHofmeister, 2022-03-20:
Dear Semu, 

Thank you for your time. I will refactor my code and take your comments into account. I will be especially mindful to CFG-VALSET and CFG-VALGET the RAM configuration first, and only secondly do the same with FLASH. 

As for the baudrate, 460800 is the recommended rate for moving-base applications according to the doumentation: https://www.u-blox.com/en/docs/UBX-19009093 

> For the UART interface a speed of 460800 baud is ideal.

Thank you for your time. This is an amazing library.

closed 2022-03-20 by dHofmeister