semuconsulting/pyubx2 #63: UnicodeDecodeError when opening ubx file

FSet89, 2022-04-13:
I downloaded an UBX file for Assisted GPS as described [here](https://developer.thingstream.io/guides/location-services/assistnow-getting-started-guide). Since I did not receive any acknowledgement from the receiver when writing it to serial, I wanted to check the content of the file. I used this code:

```
from pyubx2 import UBXReader
stream = open('mgaonline.ubx', 'rb')
UBXReader(stream, protfilter=2)
for (raw_data, parsed_data) in ubr: print(parsed_data)
```

and I got this error:
**Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/pyubx2/ubxreader.py", line 79, in __next__
    (raw_data, parsed_data) = self.read()
  File "/usr/local/lib/python3.6/dist-packages/pyubx2/ubxreader.py", line 103, in read
    byte1 = self._read_bytes(1)  # read the first byte
  File "/usr/local/lib/python3.6/dist-packages/pyubx2/ubxreader.py", line 242, in _read_bytes
    data = self._stream.read(size)
  File "/usr/lib/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb5 in position 0: invalid start byte**

semuadmin, 2022-04-14:
Hi @FSet89 AssistNow is a proprietary u-blox service and is not directly supported by pyubx2.

Having said that, to the best of my knowledge the files output by AssistNow Offline are MGA almanac and ephemeris *payloads*, not complete binary UBX messages. If you take a look at the test script in ```/tests/test_assistnow.py``` you can see an illustration of how to format MGA-GPS-ALM or MGA-GPS-EPH messages from AssistNow data for sending to the receiver.

If you want to post your mgaonline.ubx, I'm happy to take a further look.

semuadmin, 2022-04-14:
Just an aside, but Python 3.6 is now end of life (as of December 2021). We'll be dropping active support in the next major release of pyubx2.

FSet89, 2022-04-14:
Apologies, I had a bug in my code (I did not include the 'rb' parameter in `open`)

closed 2022-04-14 by FSet89

semuadmin, 2022-04-14:
That'll do it! No worries, glad it's sorted. Obviously helps to post the code you're actually using ;-)