semuconsulting/pyubx2 #64: CFG-DAT / Set Standard Datum

pdecourcel, 2022-04-22:
Using pyubx2 1.2.6

I try to build a UBXMessage from "CFG-DAT - 06 06 02 00 00 00" (this message comes from uCenter feature Tools/Receiver configuration/Transfer GNSS -> File, with a M8N PROTVER 18.0 & FWVER=SPG 3.01)

I have the following exception :
```
Traceback (most recent call last):
  File "d:\atp_holybro\env\lib\site-packages\pyubx2\ubxmessage.py", line 106, in _do_attributes
    (offset, index) = self._set_attribute(
  File "d:\atp_holybro\env\lib\site-packages\pyubx2\ubxmessage.py", line 168, in _set_attribute
    offset = self._set_attribute_single(att, offset, key, index, **kwargs)
  File "d:\atp_holybro\env\lib\site-packages\pyubx2\ubxmessage.py", line 266, in _set_attribute_single
    val = bytes2val(valb, att)
  File "d:\atp_holybro\env\lib\site-packages\pyubx2\ubxhelpers.py", line 296, in bytes2val
    val = struct.unpack("<d", valb)[0]
struct.error: unpack requires a buffer of 8 bytes

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "d:\atp_holybro\atp_holybro\ubx_config_parser.py", line 26, in <module>
    p.build_msg()
  File "d:\atp_holybro\atp_holybro\ubx_config_parser.py", line 16, in build_msg
    msg = UBXMessage(bytes[0], bytes[1], GET, payload=bytes[2:-1])
  File "d:\atp_holybro\env\lib\site-packages\pyubx2\ubxmessage.py", line 81, in __init__
    self._do_attributes(**kwargs)
  File "d:\atp_holybro\env\lib\site-packages\pyubx2\ubxmessage.py", line 117, in _do_attributes
    raise ube.UBXTypeError(
pyubx2.exceptions.UBXTypeError: Incorrect type for attribute 'majA' in GET message class CFG-DAT

```
It seems that this message is not well understood ; in older version of [protocol specification](https://content.u-blox.com/sites/default/files/products/documents/u-blox6_ReceiverDescrProtSpec_%28GPS.G6-SW-10018%29_Public.pdf) §31.3.2, a "Set Standard Datum" message is defined, with just a enumerate of Geodetic Datum.

It should be good to add the support of this CFG-DAT / Set Standard Datum message.

I would be happy to help you for developping & testing this feature.

semuadmin, 2022-04-24:
Hi @pdecourcel 

An afterthought - there is (as you've alluded to above) an older alternate definition of CFG-DAT,  but this has been obsolete since UBX protocol version 6.02 and isn't supported by Generation 7+ devices:

```
    "CFG-DAT": {
        "datumNum": U2,
    },
```

`pyubx2` doesn't currently support this obsolete version but it's easy enough to add it in _provided_ a clear distinction can be made between the two alternate versions when invoking the UBXMessage constructor. I would propose that the older, shorter version is used in the following specific circumstances:

* a keyword of `datumNum` is provided
* a `payload` with length 2 bytes is provided

e.g.

```
msg1 = UBXMessage('CFG','CFG-DAT',SET,datumNum=8)
print(msg1)
<UBX(CFG-DAT, datumNum=8)>
msg2 = UBXMessage('CFG','CFG-DAT',SET,payload=b'\x08\x00')
print(msg2)
<UBX(CFG-DAT, datumNum=8)>
```

In all other circumstances, the newer longer version would be used as per my previous reply.

Would this work for you?

closed 2022-04-24 by semuadmin

pdecourcel, 2022-04-25:
Many thanks for your replies !
It's quite curious : I recorded the receiver configuration through uCenter (22.02, 8/2/2022) menu Tools/Receiver Configuration/Transfert GNSS -> File, and I spied the COM link during this config dump (thanks to uCenter Packet Console & Binary Console) :
[image]
The receiver send a complete CFG-DAT message (60 bytes / length field = 0x34), but uCenter record it just with the old format :
`CFG-DAT - 06 06 02 00 00 00`

And of course, when I sent back this configuration to the receiver, this message is discarded... :
[image]

So it's OK for the ticket closure. It's much more a "feature" of uCenter ;-)

semuadmin, 2022-04-25:
Hi @pdecourcel 

I can only suggest you raise this on the u-blox support forum at  https://portal.u-blox.com/s/. They may be able to account for what you're seeing.  But, as I say, you are dealing with an obsolete protocol version here.

pyubx2 should now support the legacy definitions of the CFG-DAT message on paper, but unfortunately I no longer have a device to hand which is running on UBX protocol <= 6.02 so can't actually test the shorter version on a live device - all the devices I have reject it (ACK-NAK).

PR #66 (2022-04-24): Support obsolete cfg dat set datum message Fixes #64
# pyubx2 Pull Request Template

## Description

Add support for alternate versions of now-obsolete CFG-DAT SET message. There are two versions of this message type; one which takes a single unsigned integer (U2) representing a preset datum enumeration; the other takes a set of floats as manual datum settings. A modified get_dict() routine distinguishes between them by checking for a keyword of 'datumNum' or a payload length of 2 bytes.

Fixes #64 

## Testing

Please test all changes, however trivial, against the supplied unittest suite `tes [... cut]
commit 10137efd (2022-04-24): add support for obsolete CFG-DAT Fixes #64
commit b36ea30b (2022-04-24): Merge pull request #66 from semuconsulting/support-obsolete-CFG-DAT-set-datum-message

Support obsolete cfg dat set datum message Fixes #64