semuconsulting/pyubx2 #68: Wheel tick measurement support

Hothaifa, 2022-05-03:
**Is your feature request related to a problem? Please describe.**
Ublox has a number of chips that support wheel ticks like M9L, F9R and F9K. usually it is done via hardware put recently streaming wheels ticks via software is more important and easier since new wheel encoders are digital. this will eliminate the need to convert digital signal to analog signal. Also, using SW interface allow you to stream two wheels instead of one which improves performance. 

I'd like to read wheel ticks data from serial interface and feed it to the receiver.  

**Describe the solution you'd like**
being able to feed wheel ticks to F9R chip to improve performance. F9R should be able to calibrate wheel ticks and use it in DR in addition to built-in IMU

**Describe alternatives you've considered**

The alternative is to provide the wheel ticks via HW but an MCU needed to convert the digital msgs to analog. 

**Would you be willing to assist with testing?**

I have ublox F9R dev kit that I can use for testing. I can help with testing your code. I'm an armature programmer I'm not sure if I can help with writing test code. 

**Additional context**

semuadmin, 2022-05-03:
Hi @Hothaifa 

If I understand your requirement correctly, you're referring to the External Sensor Fusion (ESF) functionality built into receivers like the F9R, in which case `pyubx2` already supports this via the UBX ESF-MEAS message type.

As you're aware, there are basically two ways of feeding odometer (wheel tick) or other DR sensor data into an ESF-compatible receiver:

1. Hardware - analogue data capture via dedicated hardware pins on the receiver module (_you don't necessarily need an external MCU_).
2. Software - send an ESF-MEAS UBX message containing the relevant sensor data to the receiver via one of its available comms ports (e.g. UART1).

The ESF-MEAS SET message is defined in `ubxtypes_set.py` as follows:

```
    "ESF-MEAS": {
        "timeTag": U4,
        "flags": (
            X2,
            {
                "timeMarkSent": U2,
                "timeMarkEdge": U1,
                "calibTtagValid": U1,
                "reserved0": U7,
                "numMeas": U5,
            },
        ),
        "id": U2,
        "group": (
            "None",
            {  # repeating group * numMeas
                "data": (
                    X4,
                    {
                        "dataField": X24,
                        "dataType": U6,
                    },
                ),
            },
        ),
    },
```

The repeating `dataField` and `dataType` attributes contain the actual sensor data. You'll need to refer to the F9R Interface Specification and/or Integration Manual (_links below_) for details of how to populate these attributes and which dataTypes are supported, but once populated it's a simple matter to upload the ESF-MEAS message to the receiver - simply follow the example in the README here: https://github.com/semuconsulting/pyubx2#generating.

https://www.u-blox.com/sites/default/files/ZED-F9R_Integrationmanual_UBX-20039643.pdf
https://www.u-blox.com/sites/default/files/F9-HPS-1.21_InterfaceDescription_UBX-21019746.pdf

Hope this helps

closed 2022-05-03 by semuadmin

Hothaifa, 2022-05-06:
Thanks for your kindly reply. If possible, can you give an example on how to use/push UBX ESF-MEAS to the receiver using your library.
Regards

semuadmin, 2022-05-06:
Assuming you have the sensor data available to populate the ESF-MEAS message (_refer to the guidelines in the Integration Manual - I'm not in a position to tell you how to do this_), you can create a UBXMessage object and send it to the receiver using something like this (_substitute the PORT and BAUD values for your particular receiver configuration_):

```
>>> PORT = "COM3"
>>> BAUD = 38400
>>> from serial import Serial
>>> from pyubx2 import UBXMessage, SET
>>> serialOut = Serial(PORT, BAUD, timeout=5)
>>> msg = UBXMessage('ESF','ESF-MEAS', SET, timeTag=???, numMeas = ???, etc. etc.) # populate as per Integration Manual guidelines
>>> serialOut.write(msg.serialize())
```

Hope this helps. If you have any further queries on how to construct a suitable ESF-MEAS message, I would refer you to the u-blox support forum at https://portal.u-blox.com/s/