semuconsulting/pyubx2 #69: UNKNOWN PROTOCOL(header=b'$4')

ksiddi, 2022-05-04:
Hi

I'm trying to use the Reading example script to get te parsed data of the gnss:

>>> from serial import Serial
>>> from pyubx2 import UBXReader
>>> stream = Serial('/dev/tty**USB0**', 9600, **timeout=None**)
>>> ubr = UBXReader(stream, **portfilter=2**)
>>> (raw_data, parsed_data) = ubr.read()
>>> print(parsed_data)

You can see in bold text, what I've changed.

Instead of the parsed_data I get:
<Unkown protocol(header=b'$4')>

I know that the header of my outputs is b'\x01' and it should be b'\xb5b'

I'm using the gnss 7 click neo M9N with a serial basic adapter on a jetson nano

Is there a parameter I can change?
Thanks!

semuadmin, 2022-05-04:
Hi @ksiddi 

Sorry you're having difficulties. You haven't specified which version of pyubx2 you're using, but assuming you're using the latest version (>=1.2.7) then 'Unknown Protocol' issues are usually due to a corrupted Serial stream, typically because the serial port configuration is incorrect (e.g. baud rate, parity, etc.) or because some other process is attempting to use the same serial port (e.g. gpsd, etc).

Firstly, there is a small issue with your code - `portfilter=2` should be `protfilter=2` (_assuming your intention is to filter out all but UBX messages_). In the absence of a valid `protfilter` setting, the default is 3 (NMEA & UBX), so as things stand you would see both NMEA and UBX output.

If I understand correctly, you have a Mikroe Click 7 M9N device connected to a Jetson Nano via an SparkFun Serial Basic FTDI adapter, so I presume you're physically connected to one of the M9N's two UART ports (UART1 or UART2). Note that UBX is not normally enabled by default on UART2 so it would have to be explicitly enabled if you're using this port. The default baud rate on UART1/2 is 38400 so you would need to have explicitly configured the ports to run at 9600 to match your code. Have you done this configuration and - if so - how?

If you're sure you have the correct serial port configuration on the M9N, then it may be an issue with the Jetson-FTDI interface.  I believe the Click 7 and Jetson Nano both have USB2 ports, so have you tried connecting via USB to see if you get the same problem?

The other thing to check is if there are any other processes running on the Jetson which might be interfering with serial comms.

Try these and let me know how you get on.

ksiddi, 2022-05-05:
Thank you for answering so fast!!
I now changed the setup a bit and connected the gnss serial pins directly to the jetson and got different raw data output. 
I changed the baudrate to 38400 as you said and it works now!
Thanks again!

semuadmin, 2022-05-05:
OK great - glad the problem is sorted. I'll close this now

closed 2022-05-05 by semuadmin