semuconsulting/pyubx2 #76: ESF-MEAS message cannot input dataField and dataType

rexsham, 2022-06-15:
I used the following code and try to send a ESF-MEAS message to F9R module.
Sending messages to F9R is successful, but I failed to generate a correct message using the generator.
When I used the following code, I cannot make dataField and dataType into the generated message.

`msg = UBXMessage("ESF", "ESF-MEAS", SET , timeTag=int((round((time.time()-starttime) * 1000))), timeMarkSent=0, timeMarkEdge=0, calibTtagValid=0, numMeas=1, dataField_01=188, dataType_01=11 ) 
print("msg generated:")
print(msg)
output = msg.serialize()
print ("Parsed:")
parsed = UBXReader.parse(output)
print(parsed)`

The printed results are:
msg:
<UBX(ESF-MEAS, timeTag=6082, timeMarkSent=0, timeMarkEdge=0, calibTtagValid=0, numMeas=1, id=0)>
Parsed:
<UBX(ESF-MEAS, timeTag=6082, timeMarkSent=0, timeMarkEdge=0, calibTtagValid=0, numMeas=1, id=0)>

There is no dataField_01 or dataType_01.
Am I using a wrong syntax or is it a bug of the library?
Thanks in advance for this great library.

semuadmin, 2022-06-15:
Hi @rexsham 

Sorry you're having difficulties. This is actually 'expected behaviour' but you've got me wondering whether it needs to be. There follows a fairly long-winded explanation for why it currently works the way it does and what could be done to improve matters...

ESF-MEAS SET is a slightly unusual case. The `data` group in ESF-MEAS is currently defined as 'variable by size' - that is, it has indeterminate dimensions. The [README](https://github.com/semuconsulting/pyubx2#generating) does explain that, in such circumstances, the payload _must_ be defined using the `payload` keyword i.e. something like this:

`msg = UBXMessage("ESF", "ESF-MEAS", SET, payload=b"@B\x0f\x00\x00\x08\x00\x00\xbc\x00\x00\x0b")`

For most UBX payloads containing 'variable-by-size' repeating groups, this is a cumbersome but unavoidable solution.

However, in the case of the ESF-MEAS message there _is_ a bitfield that can _in certain circumstances_ be used to specify the data group dimension - namely `numMeas`. The specific circumstances are as follows:
- if `calibTtagValid` is 0, numMeas represents the dimension of the data group
- if `calibTtagValid` is 1, numMeas represents the dimension of the data group - 1 (_the final item in the data group is the appended calibration tag_)

So if we assume `calibTtagValid` is 0, the payload definition for ESF-MEAS could be redefined as follows:

```
    "ESF-MEAS": {
        "timeTag": U4,
        "flags": (
            X2,
            {
                "timeMarkSent": U2,
                "timeMarkEdge": U1,
                "calibTtagValid": U1,
                "reserved0": U7,
                "numMeas": U5,
            },
        ),
        "id": U2,
        "group": (
            "numMeas", # CHANGED FROM "None"
            {  # repeating group * numMeas
                "data": (
                    X4,
                    {
                        "dataField": X24,
                        "dataType": U6,
                    },
                ),
            },
        ),
    },
```

Your code would then work exactly as you expect (I've assumed a value for`starttime`:

```
msg = UBXMessage(
    "ESF",
    "ESF-MEAS",
    SET,
    timeTag=int((round((time.time() - starttime) * 1000))),
    timeMarkSent=0,
    timeMarkEdge=0,
    calibTtagValid=0,
    numMeas=1,
    dataField_01=188,
    dataType_01=11,
)

print(msg)

<UBX(ESF-MEAS, timeTag=1000000, timeMarkSent=0, timeMarkEdge=0, calibTtagValid=0, numMeas=1, id=0, dataField_01=188, dataType_01=11)>
```

So - long story short - I'm thinking I could make this the _default_ treatment for ESF-MEAS SET messages, but require users to include a final supernumerary calibration tag in the data group if `calibTtagValid =  1`. It's a trivial code change.

Does this sound like a reasonable solution to you?

rexsham, 2022-06-16:
> 

Hi @semuadmin ,
Thank you so much for your detailed reply and quick response.
I think your solution is great and it makes the usage method of this library more consistent.
This will certainly help a lot of developers developing GNSS systems with wheel tick feature.

PR #77 (2022-06-16): modify ESF-MEAS SET treatment - Fixes #76
# pyubx2 Pull Request Template

## Description

Modify handling of ESF-MEAS SET messages to use correct data group dimension based on values of `calibTtagValid` and `numMeas`. If `calibTtagValid` is True, `numMeas` is incremented by 1 to cater for the additional appended calibration `dataField` (`dataType` = 0).

Fixes #76

## Testing

Please test all changes, however trivial, against the supplied unittest suite `tests/test_*.py` e.g. by executing the `tests/testsuite.py` module or using your IDE's native Python unittest integration facilities. Please  [... cut]
commit 309f9b5c (2022-06-16): modify ESF-MEAS SET treatment - Fixes #76
commit 4342cb9d (2022-06-16): Merge pull request #77 from semuconsulting/improve-ESF-MEAS-SET

modify ESF-MEAS SET treatment - Fixes #76
closed 2022-06-16 by semuadmin