semuconsulting/pyubx2 #78: UBXMessageError: Undefined configuration database key 0x10010001

sciqader, 2022-06-29:
Hi,
I have a ZED-F9T module with which I have made a measurement via U-Center and produced a binary output log file containing both NMEA and UBX messages. I can read the NMEA messages with pyubx2 without any problems (i.e. when portfilter=1), but when I try to read the UBX messages I get the error message shown below. 

Is there a solution?

In [10]: from pyubx2 import UBXReader

In [11]: stream = open('Messung_Galileo_GPS.ubx', 'rb')

In [12]: ubr = UBXReader(stream, protfilter=2)

In [13]: for (raw_data, parsed_data) in ubr:
    ...:     print(parsed_data)
    ...: 
<UBX(MON-VER)>
<UBX(MON-VER, swVersion=b'EXT CORE 1.00 (71b20c)\x00\x00\x00\x00\x00\x00\x00\x00', hwVersion=b'00190000\x00\x00', extension_01=b'ROM BASE 0x118B2060\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', extension_02=b'FWVER=TIM 2.01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', extension_03=b'PROTVER=29.00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', extension_04=b'MOD=ZED-F9T\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', extension_05=b'GPS;GLO;GAL;BDS\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', extension_06=b'SBAS;QZSS\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')>
<UBX(CFG-VALGET, version=0, layer=0, position=0)>
---------------------------------------------------------------------------
UBXMessageError                           Traceback (most recent call last)
<ipython-input-13-cd274b995fbc> in <module>
----> 1 for (raw_data, parsed_data) in ubr:
      2     print(parsed_data)
      3 

/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyubx2/ubxreader.py in __next__(self)
     86         """
     87 
---> 88         (raw_data, parsed_data) = self.read()
     89         if raw_data is not None:
     90             return (raw_data, parsed_data)

/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyubx2/ubxreader.py in read(self)
    118                 # if it's a UBX message (b'\xb5\x62')
    119                 if bytehdr == ubt.UBX_HDR:
--> 120                     (raw_data, parsed_data) = self._parse_ubx(bytehdr)
    121                     # if protocol filter passes UBX, return message,
    122                     # otherwise discard and continue

/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyubx2/ubxreader.py in _parse_ubx(self, hdr)
    178         # only parse if we need to (filter passes UBX)
    179         if self._protfilter & ubt.UBX_PROTOCOL:
--> 180             parsed_data = self.parse(
    181                 raw_data,
    182                 validate=self._validate,

/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyubx2/ubxreader.py in parse(message, **kwargs)
    377             if payload is None:
    378                 return UBXMessage(clsid, msgid, msgmode)
--> 379             return UBXMessage(
    380                 clsid,
    381                 msgid,

/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyubx2/ubxmessage.py in __init__(self, ubxClass, ubxID, msgmode, **kwargs)
     79             self._ubxID = ubxID
     80 
---> 81         self._do_attributes(**kwargs)
     82 
     83         self._immutable = True  # once initialised, object is immutable

/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyubx2/ubxmessage.py in _do_attributes(self, **kwargs)
    104                 pdict = self._get_dict(**kwargs)  # get appropriate payload dict
    105                 for key in pdict:  # process each attribute in dict
--> 106                     (offset, index) = self._set_attribute(
    107                         offset, pdict, key, index, **kwargs
    108                     )

/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyubx2/ubxmessage.py in _set_attribute(self, offset, pdict, key, index, **kwargs)
    162                     )
    163             else:  # repeating group of attributes
--> 164                 (offset, index) = self._set_attribute_group(
    165                     att, offset, index, **kwargs
    166                 )

/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyubx2/ubxmessage.py in _set_attribute_group(self, att, offset, index, **kwargs)
    194             and self._mode == ubt.GET
    195         ):
--> 196             self._set_attribute_cfgval(offset, **kwargs)
    197         else:
    198             # derive or retrieve number of items in group

/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyubx2/ubxmessage.py in _set_attribute_cfgval(self, offset, **kwargs)
    460                     self._payload[offset : offset + KEYLEN], "little", signed=False
    461                 )
--> 462                 (keyname, att) = cfgkey2name(key)
    463                 atts = attsiz(att)
    464                 valb = self._payload[offset + KEYLEN : offset + KEYLEN + atts]

/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyubx2/ubxhelpers.py in cfgkey2name(keyID)
    374         if keyID == kid:
    375             return (key, typ)
--> 376     raise ube.UBXMessageError(f"Undefined configuration database key {hex(keyID)}")
    377 
    378 

UBXMessageError: Undefined configuration database key 0x10010001

semuadmin, 2022-06-29:
Hi @sciqader 

Can you clarify exactly how you produced this extract file and what specifically you were looking to achieve?

In addition to the NMEA output messages (which, as you say, parse OK), it appears to contain a combination of both CFG-VALGET POLL messages (i.e. queries _to_ the receiver), their ACK-ACK GET acknowledgements and their corresponding CFG-VALGET GET responses (i.e. outputs _from_ the receiver). The extract appears to contain a complete dump of every single UBX configuration database entry (_of which there are well over a thousand_) - was that your intention?

As explained in the README, for technical reasons pyubx2 defines UBX messages according to three categories - GET (outputs _from_ the receiver), SET (commands _to_ the receiver) and POLL (queries _to_ the receiver). pyubx2 can only parse one of these categories of message in a single pass, as set by the `msgmode` keyword parameter (which defaults to GET) - _either_ GET _or_ POLL _or_ SET, but not a combination of these in the same file. 

I'm not entirely sure if this is the cause of the problem you're seeing, but having a mix of POLL and GET messages in the same file will certainly cause errors. Are you able to produce an extract file containing _ONLY_ the CFG-VALGET outputs _from_ your receiver and check if that parses OK?

semuadmin, 2022-06-30:
Hi @sciqader 

I'm guessing you're using the File..Save or Record facility in u-center 21.09 to produce your .ubx dump file, with the 'Add Receiver Configuration' option set to 'Yes'. When you select Save or Record in u-center 21.09, a pop-up appears asking you whether or not you want to save the complete configuration in the logfile - this is used by u-blox for debugging purposes (the default response is 'Yes'):

[image]

If you select 'No' here, the resulting .ubx file should NOT include all the CFG-VALGET POLL/GET messages and pyubx2 should be able to parse it successfully. Try this and let me know how you get on.

semuadmin, 2022-06-30:
By the way, you may want to consider using our PyGPSClient GUI application rather than u-center to do your UBX receiver configuration and data logging. It provides quick and easy ways to enable specific UBX message types and is capable of producing log files in a variety of formats. And unlike u-center, It also runs on MacOS, Linux, Raspberry Pi as well as Windows.

https://github.com/semuconsulting/PyGPSClient

sciqader, 2022-06-30:
> Hi @sciqader
>
> I'm guessing you're using the File..Save or Record facility in u-center 21.09 to produce your .ubx dump file, with the 'Add Receiver Configuration' option set to 'Yes'. When you select [...]

Hi @semuadmin,

Thank you for your valuable suggestion. It has solved the problem! 
I will also check out the PyGPSClient.

I really appreciate your time.

semuadmin, 2022-06-30:
OK glad it's sorted.  I'll close the issue now if that's OK.

semuadmin, 2022-07-02:
Hi @sciqader 

For future reference, I have now uploaded an enhancement to the pyubx2 library (v1.2.16) which means it can now accommodate the undocumented debug message classes and configuration database keys which are present in a default u-center .ubx log file - i.e. the messages that were causing your code to fail earlier.

I would still recommend selecting 'No' to the 'Add Receiver Configuration' prompt when saving u-center .ubx files for the sake of brevity, unless your intention is to send them to u-blox technical support.

sciqader, 2022-07-04:
Hello @semuadmin 

Thank you for your message and the enhancement of the pyubx2 package. 
I have just tested the new version and can confirm that it also works for log files containing configuration database keys.

I appreciate your support very much.

Have a nice start to the week!

commit 66c175ac (2022-07-02): update to v1.2.16 Fixes #78 Fixes #79