semuconsulting/pyubx2 #80: UBXMessage throws UBXTypeError for headMot parameter

mfrydrysiak, 2022-07-07:
**Describe the bug**

UBXTypeError is thrown when providing `headMot` parameter to `UBXMessage` constructor. According to the documentation this field is `I4` with scaling `10^-5`, thus I pass there value in range `0...359` multiplied by `pow(10,5)`:

```python
pvt = UBXMessage(0x01, 0x07, GET, headMot=int(self.hom*pow(10,5)), month=6, fixType=3, gSpeed=3000, flags=b"\x01", parsebitfield=0)
```

where `self.hom` is in range `1...359`.

pyubx2 version: 1.2.16

Traceback:
```
Traceback (most recent call last):
  File "~/.local/lib/python3.8/site-packages/pyubx2/ubxmessage.py", line 106, in _do_attributes
    (offset, index) = self._set_attribute(
  File "~/.local/lib/python3.8/site-packages/pyubx2/ubxmessage.py", line 168, in _set_attribute
    offset = self._set_attribute_single(att, offset, key, index, **kwargs)
  File "~/.local/lib/python3.8/site-packages/pyubx2/ubxmessage.py", line 284, in _set_attribute_single
    valb = val2bytes(int(val / scale), att)
  File "~/.local/lib/python3.8/site-packages/pyubx2/ubxhelpers.py", line 232, in val2bytes
    valb = val.to_bytes(atts, byteorder="little", signed=True)
OverflowError: int too big to convert

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "~/test.py", line 78, in onTimerTimeout
    pvt = UBXMessage(0x01, 0x07, GET, headMot=int(self.hom*pow(10,5)), month=6, fixType=3, gSpeed=3000, flags=b"\x01", parsebitfield=0)
  File "~/.local/lib/python3.8/site-packages/pyubx2/ubxmessage.py", line 81, in __init__
    self._do_attributes(**kwargs)
  File "~/.local/lib/python3.8/site-packages/pyubx2/ubxmessage.py", line 125, in _do_attributes
    raise ube.UBXTypeError(
pyubx2.exceptions.UBXTypeError: Overflow error for attribute 'headMot' in GET message class NAV-PVT
```

**To Reproduce**

```python
UBXMessage(0x01, 0x07, GET, headMot=int(20*pow(10,5)), month=6, fixType=3, gSpeed=3000, flags=b"\x01", parsebitfield=0)
```

**Expected Behaviour**

A UBXMessage is created without throwing exception.

**Desktop (please complete the following information):**

OS: Ubuntu 20.04
Python 3.8.10

**GNSS/GPS Device (please complete the following information as best you can):**

- Device Model/Generation: u-blox ZED-F9P
- Iface: 27.31 [LINK](https://content.u-blox.com/sites/default/files/documents/u-blox-F9-HPG-1.32_InterfaceDescription_UBX-22008968.pdf)

**Additional context**

NONE

closed 2022-07-07 by mfrydrysiak

semuadmin, 2022-07-07:
Hi @mfrydrysiak ,

pyubx2 applies the scaling factors automatically during message generation and parse, so you can just pass the actual value you want, e.g.

```shell
pvt = UBXMessage(0x01, 0x07, GET, headMot=247, month=6, fixType=3, gSpeed=3000, flags=b"\x01", parsebitfield=0)
print(pvt)
<UBX(NAV-PVT, iTOW=23:59:42, year=0, month=6, day=0, hour=0, min=0, second=0, valid=b'\x00', tAcc=0, nano=0, fixType=3, flags=b'\x01', flags2=b'\x00', numSV=0, lon=0, lat=0, height=0, hMSL=0, hAcc=0, vAcc=0, velN=0, velE=0, velD=0, gSpeed=3000, headMot=247, sAcc=0, headAcc=0, pDOP=0, flags3=b'\x00\x00', reserved0=0, headVeh=0, magDec=0, magAcc=0)>
```

This would also work if you know the individual bitfield values:
```shell
pvt = UBXMessage("NAV", "NAV-PVT", GET, headMot=247, month=6, fixType=3, gSpeed=3000, gnssFixOk=1)
```

Hope this helps

mfrydrysiak, 2022-07-07:
Hi @semuadmin 

Thank you for clarification.

BR