semuconsulting/pyubx2 #81: Unable to increase the rate of navposecef stream from 1Hz to 4 Hz

madyasha, 2022-07-16:
**Describe the bug**

A clear and concise description of what the bug is.
I am trying to read messages from the ublox ZED-F9K receiver via serial communication using pyubx2. However, currently, the measurement update rate for various streams such as the navposecef stream is set to 1 Hz. Since my application involves high dynamics, I would like to increase the streaming rate to ideally 4 Hz or 20 Hz (the highest rate possible). I was able to do this in ucenter-2 but I would like to do it via serial communication. I tried setting higher rates than 1 Hz but the iTOW in messages still show that we receive these measurements at 1 Hz.

semuadmin, 2022-07-16:
Please post the FULL code you are using to set the measurement rate, along with examples of the serial output of your device BEFORE and AFTER you apply your code. Please also specify how you are physically connecting to the device (i.e. which serial port, baud rate). Thanks.

madyasha, 2022-07-17:
I am using one of the example scripts, specifically, ubxsetrates.py. I am changing the RATE value here as shown. I tried baud rates of 9600 and 115200. The serial output from my device is exactly the same.

`for (msgid, msgname) in UBX_MSGIDS.items():
            if msgid[0] == 0x01:  # NAV
                msg = UBXMessage(
                    "CFG",
                    "CFG-MSG",
                    SET,
                    msgClass=msgid[0],
                    msgID=msgid[1],
                    rateUART1=RATE,
                    rateUSB=RATE,
                )`