semuconsulting/pyubx2 #82: How to extract GPS time or higher number of digits in the iTOW field instead of UTC time?

madyasha, 2022-07-16:
I am using ublox ZED-F9K receiver for serial communication. While recording the measurements, I noticed that the time shows up as iTOW instead of GPS time probably because, in the codebase, there is a conversion happening between GPS time and UTC time.
For my application, I would like to get a higher precision time from the iTOW and ideally the GPS time directly instead of the UTC time. Is there a way to extract it?

semuadmin, 2022-07-16:
The string representation (`print(msg)`) of the iTOW field is indeed converted to UTC for readability, but the original iTOW value is available as an attribute i.e. `msg.iTOW`

madyasha, 2022-07-17:
Thank you for your response. Other than the iTOW value, could we also extract the GPS time? Additionally, this iTOW value that is made available to us, is this truncated to a few decimal places, or is it the original value?

semuadmin, 2022-07-19:
Hi @madyasha 

The iTOW attribute that pyubx2 parses as UBXMessage.iTOW is, as I've said above, the original integer value.

I recommend you download and review the Interface and Integration specifications for your ZED-F9K device as these will explain what the various attributes mean.

https://content.u-blox.com/sites/default/files/LAP120_Interfacedescription_UBX-20046191.pdf
https://content.u-blox.com/sites/default/files/ZED-F9K_Integrationmanual_UBX-20046189.pdf

See, in particular, sections 3.9.3-3.9.6 of the Integration Manual, which explain in detail what the iTOW represents and how to obtain other precision time formats.

If you have more general queries on GNSS timing and configuration on UBX devices, I recommend you ask a question on the u-blox support forum:

https://portal.u-blox.com/s/

Hope this helps

closed 2022-07-19 by semuadmin