semuconsulting/pyubx2 #83: NAV-PVT via I2C?

robomike, 2022-07-22:
Is there an example on how to poll the  NAV-PVT message via I2C ?

semuadmin, 2022-07-22:
Hi @robomike No, not explicitly.  But if you already have working I2C connectivity then it's the same UBX poll commands as if you were using USB or UART. See existing Serial examples.

closed 2022-07-22 by semuadmin

robomike, 2022-07-22:
Hi @semuadmin thank you for the response.  How do I set the i2c port and address for ubxpoller to get that NAV-PVT message?  I only see an example for UART, I'm on ubuntu with a jetson

semuadmin, 2022-07-22:
These utilities don't directly support I2C connectivity, just Serial (USB/UART). You'd have to engineer an I2C connection yourself. There are Python I2C libraries available - try a search on StackOverflow etc.