semuconsulting/pyubx2 #87: rinex output

paapu88, 2022-08-15:
Dear Developers,
Is there or is there a plan to make a u-bloc to rinex (Receiver Independent Exchange Format) converter?
Terveisin, Markus

semuadmin, 2022-08-15:
Hi @paapu88 

You're not the first person to ask, but this would involve a great deal of work and there are no current plans to support such a converter in pyubx2 unless a significant new contributor pops up. 

For the time being, the RTKCONV utility within RTKLIB is probably your best option, but this is obviously not a Python utility.

paapu88, 2022-08-15:
Thanks, I'll let you know if I'm your new contributor, have to discuss with my boss. But probably not if you think it is lots of work...

closed 2022-08-15 by paapu88

semuadmin, 2022-08-15:
Hi @paapu88 

FYI the main issue is that the complete RINEX specification includes a great deal of data which is not covered by the UBX protocol, so it would not be a simple conversion from, say, a UBX message payload to its RINEX equivalent.

You could convert individual UBX message attributes via lookup tables where there's direct correspondence, but I imagine that in itself would be of limited value.

In addition, NAV data (ephemera, almanac, etc.) requires time-series aggregation of multiple UBX messages (e.g. RXM-RAW & RXM-SFRBX) over an extended period, whereas `pyubx2` was designed from the outset to be a 'single-pass' parser i.e. one GNSS payload in, one UBX message out.

It really depends what specific functionality you'd be looking for in a RINEX 'converter'.

paapu88, 2022-08-15:
Thanks! That was a valuable piece of information for a novice.