semuconsulting/pyubx2 #88: Enabling RTCM Corrections on the ZED-F9K receivers via serial communication

madyasha, 2022-09-09:
Our research group would like to use the ZED-F9K receivers with RTCM base station corrections enabled. This is because we are currently observing significant drifts (8-10 m) in the ECEF positions in the absence of these corrections. We have been able to enable the corrections via u-center but would like to enable them via serial communication.
Are there any starter codes or examples in the repository that we can use to setup the RTCM corrections?

semuadmin, 2022-09-09:
Hi @madyasha,

Sorry but it's not at all clear what you're after here? The question is fairly nebulous.

When you say "_We have been able to enable the corrections via u-center but would like to enable them via serial communication_.", I presume you mean you have used u-center's inbuilt NTRIP (RTCM3) client facility to connect to an NTRIP caster and pass the RTCM3 messages through to your ZED-F9K device.

Are you simply after the specific UBX configuration commands that enable or disable the RTCM3 protocol on the ZED-F9K's serial ports (USB/UART1/UART2)? Or are you looking for a full-blown Python implemention of an NTRIP client?

`pyubx2` is a UBX protocol parser and configuration utility. It is not an RTCM parser or NTRIP client. There is a companion utility library `pygnssutils` which is capable of acting as an NTRIP client (via the `pyrtcm` library), and there is example code there which may be relevant to your needs:

https://github.com/semuconsulting/pygnssutils/blob/main/examples/rtk_example.py

Otherwise I would recommend you raise a general query on the u-blox technical support forum at https://portal.u-blox.com/s/.

semuadmin, 2022-09-12:
Closing if no further comments

closed 2022-09-12 by semuadmin