semuconsulting/pyubx2 #92: quitonerror option doesn't work for nmeamessage.py _do_attributes errors

dborden0, 2022-10-22:
I am using bluetooth with my GPS receiver, and I believe that is why I am getting some parsing errors.  I don't get any when hard wired. When nmeamessage.py _do_attributes raises an error, I can't figure out any way to continue operation and just ignoring the corrupt message without modifying the library. It seems like there should be a graceful way to just drop the bad data and continue.

semuadmin, 2022-10-23:
Hi @dborden0 

Could I please ask you to use the Issue template when raising issues, as advised in the README.

To help me diagnose your issue, I need to have:

- Details of the specific GNSS device you're using and how you're connecting to it.
- The specific version of `pyubx2` you're using (if not the latest 1.2.17, advise upgrading and retrying)
- The platform and operating system you're running on
- A full copy of the Python script you're executing
- A full copy of the error and traceback
- What you were hoping to achieve?

Thanks in advance

semuadmin, 2022-11-10:
Hi @dborden0,

In the absence of further information (Python script, full traceback) I can't really give you a definitive answer to your issue, but I would make the following general observations.

1. When using the `ubxreader.py` `read` method, the `quitonerror` flag will only capture unknown or unsupported protocols (i.e. something other than NMEA, UBX or RTCM3). It won't necessarily capture more general errors.
2. If you wish to capture parse, stream or message errors within an iterative read loop, `ubxreader.py` provides a dedicated `iterate` method which does allow you to use the `quitonerror` flag to specify whether such errors are ignored, logged or re-raised. See the `/examples/ubxfile.py` example by way of illustration.
3. Detection of individual corrupt NMEA, UBX or RTCM3 messages should be covered by the checksum validation flag, which is turned on by default. I recommend that checksum validation is always set to `True`.

You could of course simply add a suitable `try...except` construct around your own read loop to capture any specific exception types you wish to handle or ignore.

Hope this helps. I'm closing this issue for now.

closed 2022-11-10 by semuadmin