semuconsulting/pyubx2 #93: pynmeagps>=1.0.17 Not found

kikolobo, 2022-12-18:
For some reason pynmeagps>=1.0.17 dependency can't be found.

semuadmin, 2022-12-18:
Hi @kikolobo ,

Sorry to hear you're having difficulties, but you are using the correct installation command and pynmeagps=1.0.17 is the latest available release. This is the output I get with the same installation command:

```
➜  ~ python3 -m pip install --upgrade pyubx2                                            
Defaulting to user installation because normal site-packages is not writeable
Collecting pyubx2
  Using cached pyubx2-1.2.18-py3-none-any.whl (66 kB)
Collecting pynmeagps>=1.0.17
  Using cached pynmeagps-1.0.17-py3-none-any.whl (27 kB)
Collecting pyrtcm>=1.0.0
  Using cached pyrtcm-1.0.0-py3-none-any.whl (34 kB)
Installing collected packages: pyrtcm, pynmeagps, pyubx2
Successfully installed pynmeagps-1.0.17 pyrtcm-1.0.0 pyubx2-1.2.18
```

What happens if you try to install `pynmeagps` on its own?

```
python3 -m pip install --upgrade pynmeagps
```

semuadmin, 2022-12-20:
Hi @kikolobo - Any update on this? Happy to investigate further if you can provide the details requested above. If I don't hear back by 23/12/2022, I'll assume the problem is resolved and close this issue.

kikolobo, 2022-12-20:
Hello.

I need to get home to connect to my Jetson nano.. Il let you know tonight.  It might be one of the dependencies. Let me try to replicate it again.

semuadmin, 2022-12-26:
HI @kikolobo 

In the absence of further diagnostics, I'm going to close this issue now. 

I suspect the problem was just a temporary glitch with PyPi, but if you get any further issues with the installation, by all means raise another issue but please include a FULL TRACEBACK of any errors encountered to assist with diagnosis.

closed 2022-12-26 by semuadmin