semuconsulting/pyubx2 #101: Getting CFG-PRT to change UART speed

n8ur, 2023-01-28:
Hi --

I'm trying to change the baud rate of a NEO-6M (probably a counterfeit, so that might be the problem, but I don't have a known real one that has UART connected) and finding that no matter what I do the baud rate change doesn't take -- it ACKs the command, but still talks at the old speed, and if I poll CFG-PRT afterwards it still reports the old speed.

I originally tried creating a CFG-PRT message with only the portID and baudRate fields set.  When that didn't work, I polled CFG-PRT with parsebitfield=False, then copied the resulting fields into a new SET message with only the baudrate field changed.  That did not change the speed, either.

I'd like to avoid cut-pasting a byte object as ultimately I'd like to have flexibility in changing the port params.  I'd appreciate any suggestions about what I might be doing wrong.

Thanks!

semuadmin, 2023-01-29:
Hi @n8ur 

Can you post your _entire_ Python script you're using please, and ideally a log of the receiver output?

How are you connecting to the NEO-6M?

By way of illustration, here is a log of a sequence of CFG-PRT command I've just sent to a u-blox receiver via USB. Here I'm changing the baud rates on UART1 and UART2 (portID 1 and 2) to 115200 and 57600 respectively, from the default 38400. You can see the CFG-PRT poll responses showing the original baud rates, the ACK-ACK response to the CFG_PRT update command, and the subsequent CFG-PRT poll responses showing the updated baud rates.

As you can see, it all works fine.

Here is the Python script:

```
from serial import Serial
from pyubx2 import UBXMessage, SET

with Serial("/dev/tty.usbmodem1101", 9600, timeout=3) as stream:  # USB port

    msg1 = UBXMessage(
        "CFG",
        "CFG-PRT",
        SET,
        portID=1,
        enable=0,
        pol=0,
        pin=0,
        thres=0,
        charLen=3,
        parity=4,
        nStopBits=0,
        baudRate=115200,
        inUBX=1,
        inNMEA=1,
        outUBX=1,
        outNMEA=1,
        extendedTxTimeout=0,
    )
    print(msg1)
    print(msg1.serialize())
    msg2 = UBXMessage(
        "CFG",
        "CFG-PRT",
        SET,
        portID=2,
        enable=0,
        pol=0,
        pin=0,
        thres=0,
        charLen=3,
        parity=4,
        nStopBits=0,
        baudRate=57600,
        inUBX=1,
        inNMEA=1,
        outUBX=1,
        outNMEA=1,
        extendedTxTimeout=0,
    )
    print(msg2)
    print(msg2.serialize())

    stream.write(msg1.serialize())
    stream.write(msg2.serialize())
```

And here is the log output:

```
<UBX(CFG-PRT, portID=1, reserved0=0, enable=0, pol=0, pin=0, thres=0, charLen=3, parity=4, nStopBits=0, baudRate=38400, inUBX=1, inNMEA=1, inRTCM=0, inRTCM3=1, outUBX=1, outNMEA=1, outRTCM3=1, extendedTxTimeout=0, reserved1=0)>
000: b562 0600 1400 0100 0000 c008 0000 0096  | b'\xb5b\x06\x00\x14\x00\x01\x00\x00\x00\xc0\x08\x00\x00\x00\x96' |
016: 0000 6300 2300 0000 0000 ff30            | b'\x00\x00c\x00#\x00\x00\x00\x00\x00\xff0' |

<UBX(ACK-ACK, clsID=CFG, msgID=CFG-PRT)>
000: b562 0501 0200 0600 0e37                 | b'\xb5b\x05\x01\x02\x00\x06\x00\x0e7' |

<UBX(ACK-ACK, clsID=CFG, msgID=CFG-PRT)>
000: b562 0501 0200 0600 0e37                 | b'\xb5b\x05\x01\x02\x00\x06\x00\x0e7' |

<UBX(CFG-PRT, portID=1, reserved0=0, enable=0, pol=0, pin=0, thres=0, charLen=3, parity=4, nStopBits=0, baudRate=115200, inUBX=1, inNMEA=1, inRTCM=0, inRTCM3=1, outUBX=1, outNMEA=1, outRTCM3=1, extendedTxTimeout=0, reserved1=0)>
000: b562 0600 1400 0100 0000 c008 0000 00c2  | b'\xb5b\x06\x00\x14\x00\x01\x00\x00\x00\xc0\x08\x00\x00\x00\xc2' |
016: 0100 2300 2300 0000 0000 ec1e            | b'\x01\x00#\x00#\x00\x00\x00\x00\x00\xec\x1e' |

<UBX(ACK-ACK, clsID=CFG, msgID=CFG-PRT)>
000: b562 0501 0200 0600 0e37                 | b'\xb5b\x05\x01\x02\x00\x06\x00\x0e7' |

<UBX(CFG-PRT, portID=2, reserved0=0, enable=0, pol=0, pin=0, thres=0, charLen=3, parity=4, nStopBits=0, baudRate=38400, inUBX=1, inNMEA=0, inRTCM=0, inRTCM3=1, outUBX=0, outNMEA=0, outRTCM3=1, extendedTxTimeout=0, reserved1=0)>
000: b562 0600 1400 0200 0000 c008 0000 0096  | b'\xb5b\x06\x00\x14\x00\x02\x00\x00\x00\xc0\x08\x00\x00\x00\x96' |
016: 0000 6100 2000 0000 0000 fb22            | b'\x00\x00a\x00 \x00\x00\x00\x00\x00\xfb"' |

<UBX(ACK-ACK, clsID=CFG, msgID=CFG-PRT)>
000: b562 0501 0200 0600 0e37                 | b'\xb5b\x05\x01\x02\x00\x06\x00\x0e7' |

<UBX(ACK-ACK, clsID=CFG, msgID=CFG-PRT)>
000: b562 0501 0200 0600 0e37                 | b'\xb5b\x05\x01\x02\x00\x06\x00\x0e7' |

<UBX(CFG-PRT, portID=2, reserved0=0, enable=0, pol=0, pin=0, thres=0, charLen=3, parity=4, nStopBits=0, baudRate=57600, inUBX=1, inNMEA=0, inRTCM=0, inRTCM3=1, outUBX=0, outNMEA=0, outRTCM3=1, extendedTxTimeout=0, reserved1=0)>
000: b562 0600 1400 0200 0000 c008 0000 00e1  | b'\xb5b\x06\x00\x14\x00\x02\x00\x00\x00\xc0\x08\x00\x00\x00\xe1' |
016: 0000 2100 2000 0000 0000 065b            | b'\x00\x00!\x00 \x00\x00\x00\x00\x00\x06[' |

<UBX(ACK-ACK, clsID=CFG, msgID=CFG-PRT)>
000: b562 0501 0200 0600 0e37                 | b'\xb5b\x05\x01\x02\x00\x06\x00\x0e7' |
```

n8ur, 2023-01-29:
Thanks for getting back!  My problem probably is that I wasn't explicitly setting all the values; I wasn't sure whether that was required.  I will try fixing that, and if it doesn't work I'll extract a script that duplicates my problems (it's a multi-module with lots of other constellation-setting, etc. stuff but I can pull out a standalone baud test if needed).  I will let you know if setting all the params solves the problem; I bet it does.  If not I'll upload script and log file.  Thanks again!

semuadmin, 2023-01-29:
You may have to reset or power cycle the device first,  as setting `inUBX` to 0 will effectively block incoming UBX commands on that port.

n8ur, 2023-01-29:
It looks like it was the lack of setting all the fields explicitly that messed me up.  I tried something like that when I tried to copy the fields from polling,parsing with parsebitfields=False, then putting those fields into a SET message but most have messed something up in the process.  Setting the parsed bitfields explicitly rather than unparsed fixed whatever I had wrong.

Thanks for the help, and you can close this.  Your support of this library is really great!

n8ur, 2023-01-29:
I haven't encountered that (yet), but thanks for letting me know.  I'll 
make sure that inUBX remains set in the message.  I can also do a warm 
reset if necessary.

semuadmin, 2023-01-29:
OK glad it's sorted. I'll close this now.

closed 2023-01-29 by semuadmin