semuconsulting/pyubx2 #109: pyubx2 stop processing when there is a broken data in the ubx file.

yang-neu, 2023-04-17:
pyubx2 raises an exception and stops processing when there is broken data in the ubx file.
The error message of the exception is "'str' object has no attribute '__dict__'".

Even at the latest version 1.2.23, it still happens.

@semuadmin 
Could you please ignore the error data and continue the procedure with some messages? 

My operating system is Windows 10 and records log via U-center v22.05.
Device Model:  u-blox NEO-9M

semuadmin, 2023-04-17:
HI @yang-neu 

Please provide the FULL TRACEBACK of the error message you're getting and, if possible, a copy of the UBX file you're trying to process. Thank you.

closed 2023-04-18 by yang-neu

semuadmin, 2023-04-18:
No worries. More than happy to consider any enhancements that may make life easier.