semuconsulting/pyubx2 #156: Structure of `SIC-SEG` changed in firmware 1.50

Niederb, 2024-09-02:
The recent release of the firmware HPG 1.50 changed the structure of the `SIC-SEG` message quite a bit and the new structure is not backwards compatible to the old one. Some of the reserved parts were removed and a list of jammed frequencies was added. 
I could try to come up with a PR that supports the new structure, but I'm not sure how to do it without breaking the structure for older releases. Do you support multiple firmware versions?

Niederb, 2024-09-02:
I just realized that the version in the message was increased to `0x02`. So the two versions could be separated using the version number.

closed 2024-09-02 by semuadmin

Niederb, 2024-09-03:
Thanks a lot for the quick response and fix of the issue 👍

semuadmin, 2024-09-03:
> Thanks a lot for the quick response and fix of the issue 👍

No problem - thanks for the contribution.

In answer to your question, `pyubx2` doesn't explicitly support different u-blox protocol versions, but it can support multiple variants of the same message class which - to date - is all that's been necessary to support all protocols from Generation 6 onwards - see [ubxvariants.py](https://github.com/semuconsulting/pyubx2/blob/master/src/pyubx2/ubxvariants.py). These days, u-blox generally uses the `version` attribute to differentiate between variants, but this wasn't always the case. Previously I've had to use e.g. payload length or the existence of a specific byte pattern.

PR #158 (2024-09-02): Rc 1.2.45
# pyubx2 Pull Request Template

FIXES:

1. Fix SEC-SIGLOG payload definition - thanks to @Niederb for contribution.
1. Cater for alternate (v1, v2) versions of SEC-SIG message.

Fixes #156 

## Testing

Please test all changes, however trivial, against the supplied pytest suite `tests/test_*.py`. Please describe any test cases you have amended or added to this suite to maintain >= 99% code coverage.

If you're adding new UBX message definitions for Generation 9+ devices, please check for any corresponding configuration database updates (`ubxtypes_configdb.py`).

- [x] testSEC added t [... cut]
commit db9780dd (2024-09-02): update SEC-SIG definitions - Fixes #156