semuconsulting/pyubx2 #163: Cannot input negative speed with ESF-MEAS

FSet89, 2024-10-31:
**Describe the bug**
- Device Model/Generation: u-blox F9R
- pyubx2 version: 1.2.22

I can't send a negative speed using the ESF-MEAS message:
```
import pyubx2
pyubx2.UBXMessage("ESF", "ESF-MEAS", pyubx2.SET, timeTag=1000, numMeas=1, dataField_01=-10, dataType_01=11)
```

Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/pyubx2/ubxmessage.py", line 107, in _do_attributes
    offset, pdict, key, index, **kwargs
  File "/usr/lib/python3.7/site-packages/pyubx2/ubxmessage.py", line 165, in _set_attribute
    att, offset, index, **kwargs
  File "/usr/lib/python3.7/site-packages/pyubx2/ubxmessage.py", line 219, in _set_attribute_group
    offset, attd, key1, index, **kwargs
  File "/usr/lib/python3.7/site-packages/pyubx2/ubxmessage.py", line 157, in _set_attribute
    att, offset, index, **kwargs
  File "/usr/lib/python3.7/site-packages/pyubx2/ubxmessage.py", line 333, in _set_attribute_bitfield
    self._payload += bitfield.to_bytes(bfs, "little")
OverflowError: can't convert negative int to unsigned

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.7/site-packages/pyubx2/ubxmessage.py", line 82, in __init__
    self._do_attributes(**kwargs)
  File "/usr/lib/python3.7/site-packages/pyubx2/ubxmessage.py", line 131, in _do_attributes
    ) from err
pyubx2.exceptions.UBXTypeError: Overflow error for attribute 'group' in SET message class ESF-MEAS

semuadmin, 2024-10-31:
Hi @FSet89 

According to the [F9R Integration Manual](https://content.u-blox.com/sites/default/files/ZED-F9R_Integrationmanual_UBX-20039643.pdf), the ESF-MEAS `dataField` attribute is an **unsigned** 24-bit integer (U24), so this is expected behaviour.

[image]

I recommend having another look at the [F9R Integration Manual](https://content.u-blox.com/sites/default/files/ZED-F9R_Integrationmanual_UBX-20039643.pdf) for clarification of how to use the `dataField` and `dataType` attributes.

FSet89, 2024-10-31:
Hi, you're right, however according to the application notes the speed is signed (section 6.1.2): https://content.u-blox.com/sites/default/files/documents/ZED-F9R-GettingStarted_AppNote_UBX-22035176.pdf

semuadmin, 2024-10-31:
> Hi, you're right, however according to the application notes the speed is signed (section 6.1.2): https://content.u-blox.com/sites/default/files/documents/ZED-F9R-GettingStarted_AppNote_UBX-22035176.pdf

OK so clearly a discrepancy between different u-blox documentation sources! I'll need to raise this on the [u-blox forum](https://portal.u-blox.com/s/) for clarification. If it is in fact signed, it should be an easy fix.

semuadmin, 2024-10-31:
FYI https://portal.u-blox.com/s/question/0D5Oj00000XUqA9KAL/ubx-esfmeas-how-to-enter-ve-speed-value-in-datafield