semuconsulting/pyubx2 #172: UBXMessage.config_set() throws ValueType exception for config item of X1 type

fronders, 2024-12-27:
**Describe the bug**

UBXMessage.config_set() tries to lookup attribute types and fails because of `val2bytes()` function for `att` to `valb` conversion

Issue is caused by code on line 281:

https://github.com/semuconsulting/pyubx2/blob/1227e03f7ec7e02375114db9c87b71e161718d07/src/pyubx2/ubxhelpers.py#L265-L297

which returns integer instead of bytes in case of bitmask attribute. 

Adding `to_bytes()` conversion fixes the issue. 
This is especially critical for `X8` type, where little-endian conversion is a must. 
I don't see why `X1/2/4/8` attributes need to be treated differently from `U1/2/4/8` ones

**Info**

1. The pyubx2 version 1.2.48
2. The **complete** Python script.
```python
command = pyubx2.UBXMessage.config_set(
    layers=layers,
    transaction=pyubx2.TXN_NONE,
    cfgData=[(0x20920006,0)]
)
```
3. The error message and full traceback.
```
Traceback (most recent call last):
  File "D:\test\ublox\ublox_test.py", line 3, in <module>
    command = pyubx2.UBXMessage.config_set(
  File "D:\test\.venv\lib\site-packages\pyubx2\ubxmessage.py", line 742, in config_set
    lis = lis + keyb + valb
TypeError: can't concat int to bytes
```   

**To Reproduce**

Steps to reproduce the behaviour:
1. Try to compose any CFG-VALSET item with "X1" or "X8" type

**Desktop:**

- Windows 11 x64.
- UART1 connection via USB-UART dongle.

**GNSS/GPS Device:**

- Device Model/Generation: ublox NEO-F10N
- Firmware Version: SPG 6.00
- Protocol: 40.00

semuadmin, 2024-12-28:
Hi @fronders,

This is expected behaviour for `X` type attributes. The UBX protocol makes an intentional distinction between `U`  (unsigned integer) and `X` (byte) types, and `pyubx2` has to observe this distinction. Not all `X` type attributes can be interpreted as unsigned integers, so a `.to_bytes()` conversion would *not* be appropriate in all cases.

In this case the issue can be simply resolved by supplying the value as bytes, i.e.:

```python
import pyubx2

command = pyubx2.UBXMessage.config_set(
    layers=pyubx2.SET_LAYER_FLASH,
    transaction=pyubx2.TXN_NONE,
    cfgData=[(0x20920006, b"\x00")],
)
print(command)

```

As to why some config database values are defined as `X` when a `U` type would seem more intuitive - I'm afraid you'll have to ask u-blox that question!

fronders, 2024-12-28:
OK, fair enough, I see many X8 ones are char sequences, while X1, X2 and X4 mostly are bitmasks (which should be treated as unsigned). But at least the fact that you need to pass little-endian bytes to `cfgData` needs to be documented somewhere :)

semuadmin, 2024-12-29:
>  But at least the fact that you need to pass little-endian bytes to `cfgData` needs to be documented somewhere :) 

The 'little-endian' nature of UBX payload bytes is documented in the u-blox Interface Specifications. What you need to pass to cfgData depends on the attribute type. The type associated with each individual configuration database key is documented in the relevant Interface Specification and can be referenced via [ubxtypes_configdb.py](https://github.com/semuconsulting/pyubx2/blob/master/src/pyubx2/ubxtypes_configdb.py).

> The `val2bytes(val, att)` behavior is a bit inconsistent then, as one would expect it to return bytes. 

The `val2bytes()` method *does* always return bytes *as long as valid input arguments are provided*.

> Maybe check for incoming `val` type and raise an exception telling user to pass bytes instead of dying silently?

The `pyubx2.config_set()` function did not 'die silently' - it returned a `TypeError` which is the standard Python exception for - ahem - type errors. But I concede that the wording of the error message could be more helpful and I'm happy to update this to state explicitly which attribute 'type' it's expecting. I'll put it on the backlog and move this Issue to the Discussions Ideas and Feature Request forum.

PR #174 (2025-01-01): Rc 1.2.49
# pyubx2 Pull Request Template

## Description

1. Enhance exception handling in `val2bytes()` to explicitly validate value type e.g.
```
...
TypeError: Attribute type X001 value 0 must be <class 'bytes'>, not <class 'int'>
```

Fixes #172 #173

## Testing

Please test all changes, however trivial, against the supplied pytest suite `tests/test_*.py`. Please describe any test cases you have amended or added to this suite to maintain >= 99% code coverage.

If you're adding new UBX message definitions for Generation 9+ devices, please check for any corresponding configuration database u [... cut]